nymea/berrylan #3: The connection to WLAN was successful, but IP address not shown on a screen:

sergei, 2018-11-19:
[image]

daudt, 2018-12-27:
yep, same here. Using iOS 12.1.2 and latest Berrylan image and app.

iakashpaul, 2019-01-02:
Same here no IP being displayed, I nevertheless managed to get the IP by pinging raspberrypi.local address thx to mDNS & was able to SSH into it with IP & mDNS address. 
There could be some way for the app to connect over BT & be able to re-enter WiFi details instead of having to re-flash. I'll see if I can send a PR for that.

mzanetti, 2019-04-05:
This seems to be a issue in nymea-networkmanager where it sometimes fails to read the ip address from the system. Likely a timing-issue...

I've reported this issue in https://github.com/guh/nymea-networkmanager/issues/12

Thanks.

closed 2019-04-05 by mzanetti