nymea/berrylan #7: android app just spins when looking for wifi networks?

geeksville, 2018-12-22:
Hi,

This is a great idea for a project.  I'm happy to help debug or send in PRs (I was an android dev for my day job).

However, I'm not having luck getting it to configure my device. 

My experience from using the android app:
1. Launch app
2. See BT WLAN setup on the list of devices I can configure
3. Select that item (which causes some BT traffic to the rpi - see console log below)
4. See android app sit (forever) with the circling animation around the middle (wifi) icon in your app.
5. Was this the point where I would expect to tell the pi which wifi network name and passphrase?

(Saving bug now - will add logcat output later)

To help debug I ran your pi app directly.  I can see the android app identify it and start talking to it.  
Console output from the pi app:
```
root@rpidev:/home/pi# nymea-networkmanager --mode always -d
 I | Application: Using configuration file from: "/etc/nymea/nymea-networkmanager.conf"
 I | Application: =====================================
 I | Application: Starting nymea-networkmanager "0.3.0"
 I | Application: =====================================
 I | Application: Advertising name: "BT WLAN setup"
 I | Application: Platform name: "nymea-box"
 I | Application: Mode: Core::Mode(ModeAlways)
 I | Application: Timeout: 60
 W | NymeaService: Invalid D-Bus HardwareManager interface.
 W | NymeaService: Could not init nymea D-Bus services
 I | NetworkManager: Start the network manager.
 I | NetworkManager: Initialize network manager
 I | NetworkManager: Version: "1.6.2"
 I | NetworkManager: State changed: "NetworkManagerStateConnectedGlobal"
 I | Application: NetworkManager::NetworkManagerState(NetworkManagerStateConnectedGlobal)
 I | NetworkManager: Connectivity state changed: "Full"
 I | NetworkManager: Networking enabled
 I | Application: Networkmanager networking is now enabled
 I | NetworkManager: Wireless networking enabled
 I | Application: Networkmanager wireless networking is now enabled
 I | NetworkManager: [+] NetworkDevice("lo" - "Generic", "NetworkDeviceStateUnmanaged")
 I | NetworkManager: [+] WiredNetworkDevice("eth0", "B8:27:EB:DC:CE:72", 0 [Mb/s], true, "NetworkDeviceStateActivated") 
 I | NetworkManager: [+] WirelessNetworkDevice("wlan0", "1E:E3:F4:63:25:6D", WirelessNetworkDevice::Mode(ModeInfrastructure), 0 [Mb/s], "NetworkDeviceStateUnavailable") 
 I | NetworkManager: Settings: [+] NetworkConnection("Wired connection 1", "{8023a4b6-42f4-3525-8464-6f27686923c8}", "", "802-3-ethernet", "21.12.2018 21:39") 
 I | NetworkManager: The network manager is now available
 I | Application: Networkmanager is now available.
 I | Application: Start the bluetooth service because of "always" mode.
 I | Application: Start service
 W | NymeaService: Could not enable/disable bluetooth hardware resource. D-Bus interface not available.
 I | Application: Start bluetooth service
 I | NetworkManager: Network manager initialized successfully.
 I | Application: Start the bluetooth service because of "always" mode.
 I | Application: Start service
 W | NymeaService: Could not enable/disable bluetooth hardware resource. D-Bus interface not available.
 I | Application: Start bluetooth service
 I | BluetoothServer: -------------------------------------
 I | BluetoothServer: Starting bluetooth server
 I | BluetoothServer: -------------------------------------
 I | BluetoothServer: Local device "rpidev" "B8:27:EB:76:64:D8"
 I | BluetoothServer: Create NetworkService.
 I | BluetoothServer: NetworkService: Notify state changed "\x07"
 I | BluetoothServer: NetworkService: Notify networking enabled changed: enabled
 I | BluetoothServer: NetworkService: Notify wireless networking enabled changed: enabled
 I | BluetoothServer: Create WirelessService.
 I | BluetoothServer: Start advertising "BT WLAN setup" "B8:27:EB:76:64:D8"
 I | BluetoothServer: Controller state advertising...
 I | Application: Bluetooth server started
 I | BluetoothServer: Already running.
 I | BluetoothServer: Bluetooth host in discoverable mode.
 I | BluetoothServer: Device connected "49:96:09:F3:59:F2"
 I | BluetoothServer: Controller state connected. "" "49:96:09:F3:59:F2"
 I | Application: Bluetooth client connected
 I | BluetoothServer: WirelessService: Notify wireless state changed "\x02"
 I | BluetoothServer: WirelessService: Notify wireless mode changed "\x02"
 W | qt.bluetooth.bluez: sending error response; request: 16 handle: 51 code: 10
 W | qt.bluetooth.bluez: sending error response; request: 8 handle: 1 code: 10
 W | qt.bluetooth.bluez: sending error response; request: 8 handle: 11 code: 10
 W | qt.bluetooth.bluez: sending error response; request: 8 handle: 12 code: 10
 W | qt.bluetooth.bluez: sending error response; request: 8 handle: 20 code: 10
 W | qt.bluetooth.bluez: sending error response; request: 8 handle: 21 code: 10
 W | qt.bluetooth.bluez: sending error response; request: 8 handle: 23 code: 10
 W | qt.bluetooth.bluez: sending error response; request: 8 handle: 24 code: 10
 W | qt.bluetooth.bluez: sending error response; request: 8 handle: 37 code: 10
 W | qt.bluetooth.bluez: sending error response; request: 8 handle: 39 code: 10
 W | qt.bluetooth.bluez: sending error response; request: 8 handle: 49 code: 10
 I | BluetoothServer: NetworkService: Descriptor written "{00002902-0000-1000-8000-00805f9b34fb}" "\x01\x00"
 I | BluetoothServer: NetworkService: Descriptor written "{00002902-0000-1000-8000-00805f9b34fb}" "\x01\x00"
 I | BluetoothServer: NetworkService: Descriptor written "{00002902-0000-1000-8000-00805f9b34fb}" "\x01\x00"
 I | BluetoothServer: WirelessService: Descriptor written "{00002902-0000-1000-8000-00805f9b34fb}" "\x01\x00"
 I | BluetoothServer: WirelessService: Descriptor written "{00002902-0000-1000-8000-00805f9b34fb}" "\x01\x00"
 I | BluetoothServer: Got command stream "{\n    \"c\": 0\n}\n"
 I | BluetoothServer: WirelessService: Start streaming response data: 21 bytes
 I | BluetoothServer: WirelessService: Finished streaming response data
 I | BluetoothServer: Got command stream "{\n    \"c\": 5\n}\n"
 I | BluetoothServer: WirelessService: Execute get current connection.
 I | BluetoothServer: WirelessService: There is currently no access active accesspoint
 I | BluetoothServer: WirelessService: Start streaming response data: 53 bytes
 I | BluetoothServer: WirelessService: Finished streaming response data

```

geeksville, 2018-12-22:
Here is what I think the relevant logcat output is:

Is "WifiSetupManager: not working" a clue?

```

12-21 22:15:52.833 24568 24588 D libberrylan.so: (null):0 ((null)): BluetoothDiscovery: Stop discovering.
12-21 22:15:52.843 24568 24588 D libberrylan.so: (null):0 ((null)): BluetoothDiscovery: Discovery cancelled
12-21 22:15:53.156 24568 24588 D libberrylan.so: (null):0 ((null)): BluetoothDevice: Connected!
12-21 22:15:53.156 24568 24588 D libberrylan.so: (null):0 ((null)): BluetoothDevice: Connected to "BT WLAN setup" "B8:27:EB:76:64:D8"
12-21 22:15:53.156 24568 24588 D libberrylan.so: (null):0 ((null)): BluetoothDevice: Discovering...
12-21 22:15:53.973 24568 24588 D libberrylan.so: (null):0 ((null)): BluetoothDevice: Discovered!
12-21 22:15:53.974 24568 24588 D libberrylan.so: qrc:/main.qml:67 (expression for onConnectedChanged): qml: connectedChanged true
12-21 22:15:53.974 24568 24588 D libberrylan.so: (null):0 ((null)): WifiSetupManager: initializing
12-21 22:15:53.974 24568 24588 D libberrylan.so: (null):0 ((null)): WifiSetupManager: --> "{00001800-0000-1000-8000-00805f9b34fb}"
12-21 22:15:53.974 24568 24588 D libberrylan.so: (null):0 ((null)): WifiSetupManager: --> "{00001801-0000-1000-8000-00805f9b34fb}"
12-21 22:15:53.974 24568 24588 D libberrylan.so: (null):0 ((null)): WifiSetupManager: --> "{0000180a-0000-1000-8000-00805f9b34fb}"
12-21 22:15:53.974 24568 24588 D libberrylan.so: (null):0 ((null)): WifiSetupManager: --> "{e081fec0-f757-4449-b9c9-bfa83133f7fc}"
12-21 22:15:53.974 24568 24588 D libberrylan.so: (null):0 ((null)): WifiSetupManager: --> "{ef6d6610-b8af-49e0-9eca-ab343513641c}"
12-21 22:15:53.974 24568 24588 D libberrylan.so: (null):0 ((null)): WifiSetupManager: start discovering device information service...
12-21 22:15:54.224 24568 24588 D libberrylan.so: (null):0 ((null)): WifiSetupManager: Device information service discovered.
12-21 22:15:54.224 24568 24588 D libberrylan.so: (null):0 ((null)):     --> "Model Number String" "{00002a24-0000-1000-8000-00805f9b34fb}" "nymea-box"
12-21 22:15:54.224 24568 24588 D libberrylan.so: (null):0 ((null)):     --> "Firmware Revision String" "{00002a26-0000-1000-8000-00805f9b34fb}" "1.0.0"
12-21 22:15:54.224 24568 24588 D libberrylan.so: (null):0 ((null)):     --> "Hardware Revision String" "{00002a27-0000-1000-8000-00805f9b34fb}" "1.0.0"
12-21 22:15:54.224 24568 24588 D libberrylan.so: (null):0 ((null)):     --> "Software Revision String" "{00002a28-0000-1000-8000-00805f9b34fb}" "0.3.0"
12-21 22:15:54.224 24568 24588 D libberrylan.so: (null):0 ((null)):     --> "Manufacturer Name String" "{00002a29-0000-1000-8000-00805f9b34fb}" "guh GmbH"
12-21 22:15:54.224 24568 24588 D libberrylan.so: (null):0 ((null)): WifiSetupManager: start discovering network service...
12-21 22:15:54.644 24568 24588 D libberrylan.so: (null):0 ((null)): WifiSetupManager: Network service discovered.
12-21 22:15:54.644 24568 24588 D libberrylan.so: (null):0 ((null)):     --> "" "{ef6d6611-b8af-49e0-9eca-ab343513641c}" "\x07"
12-21 22:15:54.644 24568 24588 D libberrylan.so: (null):0 ((null)):         --> "Client Characteristic Configuration" "{00002902-0000-1000-8000-00805f9b34fb}" "\x00\x00"
12-21 22:15:54.644 24568 24588 D libberrylan.so: (null):0 ((null)):     --> "" "{ef6d6612-b8af-49e0-9eca-ab343513641c}" ""
12-21 22:15:54.644 24568 24588 D libberrylan.so: (null):0 ((null)):     --> "" "{ef6d6613-b8af-49e0-9eca-ab343513641c}" ""
12-21 22:15:54.644 24568 24588 D libberrylan.so: (null):0 ((null)):         --> "Client Characteristic Configuration" "{00002902-0000-1000-8000-00805f9b34fb}" "\x00\x00"
12-21 22:15:54.645 24568 24588 D libberrylan.so: (null):0 ((null)):     --> "" "{ef6d6614-b8af-49e0-9eca-ab343513641c}" "\x01"
12-21 22:15:54.645 24568 24588 D libberrylan.so: (null):0 ((null)):         --> "Client Characteristic Configuration" "{00002902-0000-1000-8000-00805f9b34fb}" "\x00\x00"
12-21 22:15:54.645 24568 24588 D libberrylan.so: (null):0 ((null)):     --> "" "{ef6d6615-b8af-49e0-9eca-ab343513641c}" "\x01"
12-21 22:15:54.645 24568 24588 D libberrylan.so: (null):0 ((null)):         --> "Client Characteristic Configuration" "{00002902-0000-1000-8000-00805f9b34fb}" "\x00\x00"
12-21 22:15:54.645 24568 24588 D libberrylan.so: (null):0 ((null)): Enable notifications of network service
12-21 22:15:54.650 24568 24588 D libberrylan.so: (null):0 ((null)): --> WirelessSetupManager::NetworkStatus(NetworkStatusGlobal)
12-21 22:15:54.650 24568 24588 D libberrylan.so: qrc:/main.qml:74 (expression for onNetworkStatusChanged): qml: Network status changed: 7
12-21 22:15:54.651 24568 24588 D libberrylan.so: (null):0 ((null)): WifiSetupManager: Networking enabled changed true
12-21 22:15:54.651 24568 24588 D libberrylan.so: (null):0 ((null)): WifiSetupManager: Wireless enabled changed true
12-21 22:15:54.651 24568 24588 D libberrylan.so: (null):0 ((null)): WifiSetupManager: start discovering wifi service...
12-21 22:15:55.126 24568 24588 D libberrylan.so: (null):0 ((null)): WifiSetupManager: Wifi service discovered.
12-21 22:15:55.127 24568 24588 D libberrylan.so: (null):0 ((null)):     --> "" "{e081fec1-f757-4449-b9c9-bfa83133f7fc}" ""
12-21 22:15:55.127 24568 24588 D libberrylan.so: (null):0 ((null)):     --> "" "{e081fec2-f757-4449-b9c9-bfa83133f7fc}" ""
12-21 22:15:55.127 24568 24588 D libberrylan.so: (null):0 ((null)):         --> "Client Characteristic Configuration" "{00002902-0000-1000-8000-00805f9b34fb}" "\x00\x00"
[... cut ...]
12-21 22:15:55.366 24568 24588 D libberrylan.so: }
12-21 22:15:55.366 24568 24588 D libberrylan.so: 
12-21 22:15:55.366 24568 24588 D libberrylan.so: (null):0 ((null)): WifiSetupManager: not working
12-21 22:15:55.366 24568 24588 D libberrylan.so: (null):0 ((null)): Current network connection QMap(("c", QVariant(double, 5))("p", QVariant(QVariantMap, QMap(("e", QVariant(QString, ""))("i", QVariant(QString, ""))("m", QVariant(QString, ""))("p", QVariant(double, 0))("s", QVariant(double, 0)))))("r", QVariant(double, 0)))
```

t-mon, 2018-12-25:
Hi geeksville! Thanks for your feedback! 
It looks like the spinner is endless running because the list of wireless networks is empty. Are you able to scan wifi networks using `nmcli device wifi`?

PS: please make sure you are not sharing any wifi credentials with the logs :-)

geeksville, 2018-12-25:
Hi @t-mon,

Thanks for your reply.  Oops I had missed uninstalling dhcpcd5.  It now reaches the point in the android app where it says "Connecting the Raspberry Pi to geeksville (my wifi)".  However, the pi side never connects to the wifi.  If I use "nmcli device wifi connect geeksville password XXX" manually I can connect to the wifi network.

Is the problem related to this message?

 W | NetworkManager: "org.freedesktop.NetworkManager.Settings.Failed" "No plugin supported adding this connection"

Here's the relevant logs from the pi:

```
root@rpidev:/home/pi# nymea-networkmanager --mode always -d
 I | Application: Using configuration file from: "/etc/nymea/nymea-networkmanager.conf"
 I | Application: =====================================
 I | Application: Starting nymea-networkmanager "0.3.0"
 I | Application: =====================================
 I | Application: Advertising name: "BT WLAN setup"
 I | Application: Platform name: "nymea-box"
 I | Application: Mode: Core::Mode(ModeAlways)
 I | Application: Timeout: 60
 W | NymeaService: Invalid D-Bus HardwareManager interface.
 W | NymeaService: Could not init nymea D-Bus services
 I | NetworkManager: Start the network manager.
 I | NetworkManager: Initialize network manager
 I | NetworkManager: Version: "1.6.2"
 I | NetworkManager: State changed: "NetworkManagerStateConnectedGlobal"
 I | Application: NetworkManager::NetworkManagerState(NetworkManagerStateConnectedGlobal)
 I | NetworkManager: Connectivity state changed: "Full"
 I | NetworkManager: Networking enabled
 I | Application: Networkmanager networking is now enabled
 I | NetworkManager: Wireless networking enabled
 I | Application: Networkmanager wireless networking is now enabled
 I | NetworkManager: [+] NetworkDevice("lo" - "Generic", "NetworkDeviceStateUnmanaged")
 I | NetworkManager: [+] WiredNetworkDevice("eth0", "B8:27:EB:DC:CE:72", 0 [Mb/s], true, "NetworkDeviceStateActivated") 
 I | NetworkManager: [+] WirelessNetworkDevice("wlan0", "AA:C8:D8:DE:C6:DB", WirelessNetworkDevice::Mode(ModeInfrastructure), 0 [Mb/s], "NetworkDeviceStateDisconnected") 
 I | NetworkManager: Settings: [+] NetworkConnection("Wired connection 1", "{8023a4b6-42f4-3525-8464-6f27686923c8}", "", "802-3-ethernet", "03.11.2016 10:16") 
 I | NetworkManager: The network manager is now available
 I | Application: Networkmanager is now available.
 I | Application: Start the bluetooth service because of "always" mode.
 I | Application: Start service
 W | NymeaService: Could not enable/disable bluetooth hardware resource. D-Bus interface not available.
 I | Application: Start bluetooth service
 I | NetworkManager: Network manager initialized successfully.
 I | Application: Start the bluetooth service because of "always" mode.
 I | Application: Start service
 W | NymeaService: Could not enable/disable bluetooth hardware resource. D-Bus interface not available.
 I | Application: Start bluetooth service
 I | BluetoothServer: -------------------------------------
 I | BluetoothServer: Starting bluetooth server
 I | BluetoothServer: -------------------------------------
 I | BluetoothServer: Local device "rpidev" "B8:27:EB:76:64:D8"
 I | BluetoothServer: Create NetworkService.
 I | BluetoothServer: NetworkService: Notify state changed "\x07"
 I | BluetoothServer: NetworkService: Notify networking enabled changed: enabled
 I | BluetoothServer: NetworkService: Notify wireless networking enabled changed: enabled
 I | BluetoothServer: Create WirelessService.
 I | BluetoothServer: Start advertising "BT WLAN setup" "B8:27:EB:76:64:D8"
 I | BluetoothServer: Controller state advertising...
 I | Application: Bluetooth server started
 I | BluetoothServer: Already running.
 I | BluetoothServer: Bluetooth host in discoverable mode.
 I | BluetoothServer: Device connected "7A:5B:81:AF:B2:45"
 I | BluetoothServer: Controller state connected. "" "7A:5B:81:AF:B2:45"
 I | Application: Bluetooth client connected
 I | BluetoothServer: WirelessService: Notify wireless state changed "\x03"
 I | BluetoothServer: WirelessService: Notify wireless mode changed "\x02"
 W | qt.bluetooth.bluez: sending error response; request: 16 handle: 51 code: 10
 W | qt.bluetooth.bluez: sending error response; request: 8 handle: 1 code: 10
 W | qt.bluetooth.bluez: sending error response; request: 8 handle: 11 code: 10
 W | qt.bluetooth.bluez: sending error response; request: 8 handle: 12 code: 10
 W | qt.bluetooth.bluez: sending error response; request: 8 handle: 20 code: 10
 W | qt.bluetooth.bluez: sending error response; request: 8 handle: 21 code: 10
 W | qt.bluetooth.bluez: sending error response; request: 8 handle: 23 code: 10
 W | qt.bluetooth.bluez: sending error response; request: 8 handle: 24 code: 10
 W | qt.bluetooth.bluez: sending error response; request: 8 handle: 37 code: 10
 W | qt.bluetooth.bluez: sending error response; request: 8 handle: 39 code: 10
 W | qt.bluetooth.bluez: sending error response; request: 8 handle: 49 code: 10
 I | BluetoothServer: NetworkService: Descriptor written "{00002902-0000-1000-8000-00805f9b34fb}" "\x01\x00"
 I | BluetoothServer: NetworkService: Descriptor written "{00002902-0000-1000-8000-00805f9b34fb}" "\x01\x00"
 I | BluetoothServer: NetworkService: Descriptor written "{00002902-0000-1000-8000-00805f9b34fb}" "\x01\x00"
 I | BluetoothServer: WirelessService: Descriptor written "{00002902-0000-1000-8000-00805f9b34fb}" "\x01\x00"
[... cut ...]
 I | BluetoothServer: WirelessService: There is currently no access active accesspoint
 I | BluetoothServer: WirelessService: Start streaming response data: 53 bytes
 I | BluetoothServer: WirelessService: Finished streaming response data
 I | BluetoothServer: Got command stream "{\n    \"c\": 1,\n    \"p\": {\n        \"e\": \"geeksville\",\n        \"p\": \"deadbeef01\"\n    }\n}\n"
 W | NetworkManager: "org.freedesktop.NetworkManager.Settings.Failed" "No plugin supported adding this connection"
```

andreialecu, 2019-01-06:
Had the same issue until I ran:

apt-get remove dhcpcd5

The command in the FAQ to disable dhcpd didn't work on my Raspbian for some reason.

geeksville, 2019-01-06:
Though to be clear my current issue occurs even after I removed dhcpcd5.

mzanetti, 2019-11-07:
We've released some fixes in the meantime. Please test and reopen if it's still an issue.

closed 2019-11-07 by mzanetti

krissyKris, 2020-02-29:
Hi, I have the same issue on iPhone version of Berrylan - it is stuck on wifi screen searching screen (spinning icon). Is removal of dhcpcd5 on RPi Zero required then (as suggested above)?

mzanetti, 2020-02-29:
Yes, dhcpcd5 needs to be removed as it clashes with NetworkManager's own DCHP feature.

krissyKris, 2020-03-01:
great, thanks for a prompt response!