nymea/berrylan #9: Bluetooth discoverable even if connected to Wifi

andreialecu, 2019-01-06:
My Pi is connected to the Wifi, and the network works properly, but the Berrylan app on the phone still sees the Pi and allows me to connect it to a different network.

I have only changed the advertised device name from the default, but it was acting the same even before this, here's the log

```
pi@raspberrypi:~ $ systemctl status nymea-networkmanager
● nymea-networkmanager.service - Daemon for nymea to configure wifi network using a Bluetooth LE connection
   Loaded: loaded (/lib/systemd/system/nymea-networkmanager.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2019-01-06 18:42:52 EET; 8min ago
     Docs: https://github.com/guh/nymea-networkmanager
 Main PID: 400 (nymea-networkma)
   CGroup: /system.slice/nymea-networkmanager.service
           └─400 /usr/bin/nymea-networkmanager -d -a yyy WIFI Setup -m offline

ian 06 18:43:00 raspberrypi nymea-networkmanager[400]:  I | BluetoothServer: NetworkService: Notify state changed "\x07"
ian 06 18:43:00 raspberrypi nymea-networkmanager[400]:  I | BluetoothServer: NetworkService: Notify networking enabled changed: enabled
ian 06 18:43:00 raspberrypi nymea-networkmanager[400]:  I | BluetoothServer: NetworkService: Notify wireless networking enabled changed: enabled
ian 06 18:43:00 raspberrypi nymea-networkmanager[400]:  I | BluetoothServer: Create WirelessService.
ian 06 18:43:00 raspberrypi nymea-networkmanager[400]:  I | BluetoothServer: Start advertising "yyy WIFI Setup" "B8:27:EB:50:16:10"
ian 06 18:43:00 raspberrypi nymea-networkmanager[400]:  I | BluetoothServer: Controller state advertising...
ian 06 18:43:00 raspberrypi nymea-networkmanager[400]:  I | Application: Bluetooth server started
ian 06 18:43:00 raspberrypi nymea-networkmanager[400]:  I | BluetoothServer: Already running.
ian 06 18:43:00 raspberrypi nymea-networkmanager[400]:  I | BluetoothServer: Bluetooth host in discoverable mode.
ian 06 18:46:23 raspberrypi nymea-networkmanager[400]:  I | BluetoothServer: Bluetooth host in connectable mode.
```

Any ideas what might be wrong? This is a custom setup on a default Raspbian install, via the steps in the FAQ.

andreialecu, 2019-01-06:
Curiously, if I re-do the setup using the app, it disappears. But it reappears on the next reboot.

andreialecu, 2019-01-06:
Here's a more complete log:

```
ian 06 18:42:52 raspberrypi systemd[1]: Started Daemon for nymea to configure wifi network using a Bluetooth LE connection.
ian 06 18:42:53 raspberrypi nymea-networkmanager[400]:  I | Application: Using configuration file from: "/etc/nymea/nymea-networkmanager.conf"
ian 06 18:42:53 raspberrypi nymea-networkmanager[400]:  I | Application: Fooo ("AdvertiseName", "Mode", "PlatformName", "Timeout") ()
ian 06 18:42:53 raspberrypi nymea-networkmanager[400]:  I | Application: =====================================
ian 06 18:42:53 raspberrypi nymea-networkmanager[400]:  I | Application: Starting nymea-networkmanager "0.2.0"
ian 06 18:42:53 raspberrypi nymea-networkmanager[400]:  I | Application: =====================================
ian 06 18:42:53 raspberrypi nymea-networkmanager[400]:  I | Application: Advertising name: "yyy WIFI Setup"
ian 06 18:42:53 raspberrypi nymea-networkmanager[400]:  I | Application: Platform name: "nymea-box"
ian 06 18:42:53 raspberrypi nymea-networkmanager[400]:  I | Application: Mode: Core::Mode(ModeOffline)
ian 06 18:42:53 raspberrypi nymea-networkmanager[400]:  I | Application: Timeout: 60
ian 06 18:42:53 raspberrypi nymea-networkmanager[400]:  W | NymeaService: Invalid D-Bus HardwareManager interface.
ian 06 18:42:53 raspberrypi nymea-networkmanager[400]:  W | NymeaService: Could not init nymea D-Bus services
ian 06 18:42:53 raspberrypi nymea-networkmanager[400]:  I | NetworkManager: Start the network manager.
ian 06 18:42:53 raspberrypi nymea-networkmanager[400]:  I | NetworkManager: Initialize network manager
ian 06 18:42:54 raspberrypi nymea-networkmanager[400]:  I | NetworkManager: Version: "1.6.2"
ian 06 18:42:54 raspberrypi nymea-networkmanager[400]:  I | NetworkManager: State changed: "NetworkManagerStateDisconnected"
ian 06 18:42:54 raspberrypi nymea-networkmanager[400]:  I | Application: NetworkManager::NetworkManagerState(NetworkManagerStateDisconnected)
ian 06 18:42:54 raspberrypi nymea-networkmanager[400]:  I | Application: Start the bluetooth service because of "offline" mode.
ian 06 18:42:54 raspberrypi nymea-networkmanager[400]:  W | Application: Could not start services. There is no network manager available.
ian 06 18:42:54 raspberrypi nymea-networkmanager[400]:  I | NetworkManager: Connectivity state changed: "None"
ian 06 18:42:54 raspberrypi nymea-networkmanager[400]:  I | NetworkManager: Networking enabled
ian 06 18:42:54 raspberrypi nymea-networkmanager[400]:  I | Application: Networkmanager networking is now enabled
ian 06 18:42:54 raspberrypi nymea-networkmanager[400]:  I | Application: Start the bluetooth service because of "offline" mode.
ian 06 18:42:54 raspberrypi nymea-networkmanager[400]:  W | Application: Could not start services. There is no network manager available.
ian 06 18:42:54 raspberrypi nymea-networkmanager[400]:  I | NetworkManager: Wireless networking enabled
ian 06 18:42:54 raspberrypi nymea-networkmanager[400]:  I | Application: Networkmanager wireless networking is now enabled
ian 06 18:42:54 raspberrypi nymea-networkmanager[400]:  I | Application: Start the bluetooth service because of "offline" mode.
ian 06 18:42:54 raspberrypi nymea-networkmanager[400]:  W | Application: Could not start services. There is no network manager available.
ian 06 18:42:54 raspberrypi nymea-networkmanager[400]:  I | NetworkManager: [+] NetworkDevice("lo" - "Generic", "NetworkDeviceStateUnmanaged")
ian 06 18:42:54 raspberrypi nymea-networkmanager[400]:  I | NetworkManager: [+] WiredNetworkDevice("eth0", "B8:27:EB:FA:BC:BA", 0 [Mb/s], false, "NetworkDeviceStateUnavailable")
ian 06 18:42:54 raspberrypi nymea-networkmanager[400]:  I | NetworkManager: [+] WirelessNetworkDevice("wlan0", "6A:F0:F2:97:C6:61", 0 [Mb/s], "NetworkDeviceStateUnavailable")
ian 06 18:42:54 raspberrypi nymea-networkmanager[400]:  I | NetworkManager: Settings: [+] NetworkConnection("A1", "{a404dece-ec2a-4b9f-a00d-0794134cd679}", "", "802-11-wireless", "06.01.2019 18:42")
ian 06 18:42:54 raspberrypi nymea-networkmanager[400]:  I | NetworkManager: Settings: [+] NetworkConnection("Wired connection 1", "{29d99b24-df93-3251-a4f6-d9a84bd1551f}", "", "802-3-ethernet", "06.01.2019 18:42")
ian 06 18:42:54 raspberrypi nymea-networkmanager[400]:  I | NetworkManager: The network manager is now available
ian 06 18:42:54 raspberrypi nymea-networkmanager[400]:  I | Application: Networkmanager is now available.
ian 06 18:42:54 raspberrypi nymea-networkmanager[400]:  I | Application: Start the bluetooth service because of "offline" mode.
ian 06 18:42:54 raspberrypi nymea-networkmanager[400]:  I | Application: Start service
ian 06 18:42:54 raspberrypi nymea-networkmanager[400]:  W | NymeaService: Could not enable/disable bluetooth hardware resource. D-Bus interface not available.
ian 06 18:42:54 raspberrypi nymea-networkmanager[400]:  I | Application: Start bluetooth service
ian 06 18:42:54 raspberrypi nymea-networkmanager[400]:  I | NetworkManager: Network manager initialized successfully.
ian 06 18:42:54 raspberrypi nymea-networkmanager[400]:  I | Application: Start the bluetooth service because of "offline" mode.
ian 06 18:42:54 raspberrypi nymea-networkmanager[400]:  I | Application: Start service
ian 06 18:42:54 raspberrypi nymea-networkmanager[400]:  W | NymeaService: Could not enable/disable bluetooth hardware resource. D-Bus interface not available.
ian 06 18:42:54 raspberrypi nymea-networkmanager[400]:  I | Application: Start bluetooth service
[... cut ...]
ian 06 18:43:00 raspberrypi nymea-networkmanager[400]:  I | BluetoothServer: Start advertising "yyy WIFI Setup" "B8:27:EB:50:16:10"
ian 06 18:43:00 raspberrypi nymea-networkmanager[400]:  I | BluetoothServer: Controller state advertising...
ian 06 18:43:00 raspberrypi nymea-networkmanager[400]:  I | Application: Bluetooth server started
ian 06 18:43:00 raspberrypi nymea-networkmanager[400]:  I | BluetoothServer: Already running.
ian 06 18:43:00 raspberrypi nymea-networkmanager[400]:  I | BluetoothServer: Bluetooth host in discoverable mode.
```

PeterFucci, 2019-01-10:
I'm seeing this issue also.  I did connect eth0 and made user wlan was working (and confirmed by accessing device on both IP's). I rebooted, and the device is still broadcasting.

andreialecu, 2019-01-10:
This is a pretty big security issue since it's allowing anyone to connect the device to any network, so it's kinda blocking my implementation.

Not sure if it's a misconfiguration issue or something else.

t-mon, 2019-01-10:
This issue should be closed with the new `nymea-networkmanager` version `0.3.2`. Please check it out and tell me if it works for you. :) Thanks

andreialecu, 2019-01-11:
Confirmed that it works properly now on the new version! Thanks so much!

closed 2019-01-11 by andreialecu

PeterFucci, 2019-01-11:
> Confirmed that it works properly now on the new version! Thanks so much!

How did you pull that specific version?
I'm used to using a SVN... this is a whole new world.!

PeterFucci, 2019-01-13:
Thanks t-mon!