nymea/berrylan #10: Install on full Raspian doesn't work

rnehrboss, 2019-01-24:
Hi.. Tested your image with raspbian lite.. works great.  Nice job.
Want image with desktop so got a fresh Raspbian Stretch full.   Followed this:
Installation on a plain Raspbian
sudo apt-get install dirmngr
echo "deb http://repository.nymea.io stretch main" | sudo tee /etc/apt/sources.list.d/nymea.list
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-key A1A19ED6
sudo apt-get update
sudo apt-get install nymea-networkmanager
sudo systemctl disable dhcpcd
reboot

Everything installed, but doesn't work.    Pi Zero is not connected to wifi (actually after insalling the above, there is no interface).. and the Android App does not find it as a bluetooth device to set the WLAN.

Anyone have any thoughts?

t-mon, 2019-01-24:
Hi @rnehrboss 

the main section of the repository holds only real armv7 packages. The raspberry pi zero is actually an armv6 showing up as armv7. For that reason we have introduced the section `raspbian`, which hold packages also working for the Pi zero. 

You could try following

    nano /etc/apt/sources.list.d/nymea.list

Change the line to following

    deb http://repository.nymea.io stretch main raspbian

Configure apt to prefere packages from rasbian section

    nano /etc/apt/preferences.d/nymea

And add following configuration

    Package: *
    Pin: release c=raspbian
    Pin-Priority: 700
    
    Package: *
    Pin: origin repository.nymea.io c=main
    Pin-Priority: 500

    apt update
    apt remove nymea-networkmanager
    apt install nymea-networkmanager

Now you should have installed the version from the `raspbian` section and it should work.

     0.3.2 700
        700 http://repository.nymea.io stretch/raspbian armhf Packages
        100 /var/lib/dpkg/status

I hope this helps, I have to add this to the documentation. Thanks for pointing it out!

rnehrboss, 2019-01-25:
Awesome response.   Pi Zero works like a charm now.  Thanks!