nymea/berrylan #15: Modify Spinner Color Scheme

sparkhead95, 2019-04-05:
I've been racking my brain for a while trying to find where the colours for the red spinner around the icons are? And the same red when you click on each icon (bluetooth, wifi etc) like below?
[image]

Also, I'm fairly new to QT but it from my Googling it seems that the app icon is not defined in any of the normal QT project locations? Where is this app icon defined?

Thanks guys

sparkhead95, 2019-04-05:
Furthermore, and more of an actual 'issue', I seem to be getting a couple of null pointers upon compilation, straight out of the box:
`qrc:/main.qml:241: TypeError: Cannot read property 'accessPointModeAvailable' of null
qrc:/main.qml:350: TypeError: Cannot read property 'currentConnection' of null`

mzanetti, 2019-04-05:
The app icon is set in the packaging, depending on the platform you're building for.

About those TypeError warnings, might be that I didn't silence them correctly... Not really an issue as QML will update itself when it becomes available once the Bluetooth connection is established. Could be silenced with changing line 241 to

`visible: networkManager.manager !== null && networkManager.manager.accessPointModeAvailable`

Also, for generic questions, please use the forum at https://forum.nymea.io (There is a berrylan section available).

Br,
Michael

mzanetti, 2019-04-05:
Closing this as it is not an actual issue

closed 2019-04-05 by mzanetti