nymea/berrylan #17: Bluetooth advertising after connected

rnehrboss, 2019-04-12:
Referring to issue #9.   I'm having the same issue (Bluetooth is advertising even after connected to wifi (somewhat intermittently).

I read through issue #9 and checked version:
nymea-networkmanager/unknown,now 0.3.2 armhf [installed]

This is on PI Zero W if that helps.

Any thoughts?

Thanks!

t-mon, 2019-04-15:
Hi @rnehrboss! Could you provide me following information here:

1. The package information:
`apt policy nymea-networkmanager`

2. The logs after a clean restart from the service:
`systemctl restart nymea-networkmanager && journalctl -f -u nymea-networkmanager` 

I hope this gives us the necessary information in order to finding the issue. 
Thanks.

rnehrboss, 2019-04-16:
Thanks for helping look into this:

pi@raspberrypi:~ $ apt policy nymea-networkmanager
nymea-networkmanager:
  Installed: 0.3.2
  Candidate: 0.3.2
  Version table:
     0.3.2+201901111109~03bff81~stretch 500
        500 http://repository.nymea.io stretch/main armhf Packages
 *** 0.3.2 700
        700 http://repository.nymea.io stretch/raspbian armhf Packages
        100 /var/lib/dpkg/status

------------------------
.
pi@raspberrypi:~ $ sudo systemctl restart nymea-networkmanager &&
journalctl -f -u nymea-networkmanager
[sudo] password for pi:
-- Logs begin at Mon 2019-04-15 20:43:07 UTC. --
Apr 16 22:02:22 raspberrypi nymea-networkmanager[3598]:  I | Application:
Shutting down nymea-networkmanager
Apr 16 22:02:22 raspberrypi nymea-networkmanager[3598]:  I | Application:
=====================================
Apr 16 22:02:22 raspberrypi nymea-networkmanager[3598]:  I | Application:
Shutting down nymea service
Apr 16 22:02:22 raspberrypi nymea-networkmanager[3598]:  I | Application:
Shutting down bluetooth service
Apr 16 22:02:22 raspberrypi nymea-networkmanager[3598]:  I |
BluetoothServer: Destroy bluetooth server.
Apr 16 22:02:22 raspberrypi nymea-networkmanager[3598]:  I | Application:
Shutting down network-manager service
Apr 16 22:02:22 raspberrypi nymea-networkmanager[3598]:  I |
NetworkManager: Destroy network manager.
Apr 16 22:02:22 raspberrypi systemd[1]: Stopping Daemon for nymea to
configure wifi network using a Bluetooth LE connection...
Apr 16 22:02:22 raspberrypi systemd[1]: Stopped Daemon for nymea to
configure wifi network using a Bluetooth LE connection.
Apr 16 22:02:22 raspberrypi systemd[1]: Started Daemon for nymea to
configure wifi network using a Bluetooth LE connection.
Apr 16 22:02:22 raspberrypi nymea-networkmanager[7940]:  I | Application:
Using configuration file from: "/etc/nymea/nymea-networkmanager.conf"
Apr 16 22:02:22 raspberrypi nymea-networkmanager[7940]:  I | Application:
=====================================
Apr 16 22:02:22 raspberrypi nymea-networkmanager[7940]:  I | Application:
Starting nymea-networkmanager "0.3.0"
Apr 16 22:02:22 raspberrypi nymea-networkmanager[7940]:  I | Application:
=====================================
Apr 16 22:02:22 raspberrypi nymea-networkmanager[7940]:  I | Application:
Advertising name: "Intelligent Demographics"
Apr 16 22:02:22 raspberrypi nymea-networkmanager[7940]:  I | Application:
Platform name: "nymea-box"
Apr 16 22:02:22 raspberrypi nymea-networkmanager[7940]:  I | Application:
Mode: Core::Mode(ModeOffline)
Apr 16 22:02:22 raspberrypi nymea-networkmanager[7940]:  I | Application:
Timeout: 60
Apr 16 22:02:23 raspberrypi nymea-networkmanager[7940]:  W | NymeaService:
Invalid D-Bus HardwareManager interface.
Apr 16 22:02:23 raspberrypi nymea-networkmanager[7940]:  W | NymeaService:
Could not init nymea D-Bus services
Apr 16 22:02:23 raspberrypi nymea-networkmanager[7940]:  I |
NetworkManager: Start the network manager.
Apr 16 22:02:23 raspberrypi nymea-networkmanager[7940]:  I |
NetworkManager: Initialize network manager
Apr 16 22:02:23 raspberrypi nymea-networkmanager[7940]:  I |
NetworkManager: Version: "1.6.2"
Apr 16 22:02:23 raspberrypi nymea-networkmanager[7940]:  I |
NetworkManager: State changed: "NetworkManagerStateConnectedGlobal"
Apr 16 22:02:23 raspberrypi nymea-networkmanager[7940]:  I | Application:
NetworkManager::NetworkManagerState(NetworkManagerStateConnectedGlobal)
Apr 16 22:02:23 raspberrypi nymea-networkmanager[7940]:  I |
[... 39 lines cut ...]
Apr 16 22:02:24 raspberrypi nymea-networkmanager[7940]:  I |
NetworkManager: Settings: [+] NetworkConnection("upstairs 3",
"{d6df8331-4311-4e1f-9353-68ed0ee1c7e1}", "", "802-11-wireless",
"08.02.2019 19:39")
Apr 16 22:02:24 raspberrypi nymea-networkmanager[7940]:  I |
NetworkManager: Settings: [+] NetworkConnection("TP-LINK2",
"{618c63ee-3abf-4cee-8945-eb09b6d9681c}", "", "802-11-wireless",
"22.02.2019 17:26")
Apr 16 22:02:25 raspberrypi nymea-networkmanager[7940]:  I |
NetworkManager: Settings: [+] NetworkConnection("TP-LINK7",
"{a857bcd1-c34a-4240-8958-56eedb7bf3e9}", "", "802-11-wireless",
"16.04.2019 22:00")
Apr 16 22:02:25 raspberrypi nymea-networkmanager[7940]:  I |
NetworkManager: The network manager is now available
Apr 16 22:02:25 raspberrypi nymea-networkmanager[7940]:  I | Application:
Init is running
Apr 16 22:02:25 raspberrypi nymea-networkmanager[7940]:  I |
NetworkManager: Network manager initialized successfully.

Symptoms:   Sometimes the settings don't "stick".. IE have to run android
Berrylan more than once.   Often bluetooth stays on advertising even when
connected to WIFI.  Kind of a security issue.

Thanks again!
Richard

t-mon, 2019-04-23:
From what I see the version should be the fixed one, interesting. 
Can you past me also the log part where the Bluetooth server starts? The logs show only the part until the network-manager initialization (recognizing you are online), but the interesting part would be the reason why the Bluetooth server starts. Are you always able to connect to the service using the app?

> Please make sure you are not pasting any passwords with the logs ;-)

rnehrboss, 2019-04-25:
Thanks for looking into this.

It happens intermittently.  For example it is not advertising now.   Here
is the log:

Apr 25 03:38:23 raspberrypi nymea-networkmanager[4875]:  I | Application:
Shutting down nymea-networkmanager
Apr 25 03:38:23 raspberrypi nymea-networkmanager[4875]:  I | Application:
=====================================
Apr 25 03:38:23 raspberrypi nymea-networkmanager[4875]:  I | Application:
Shutting down nymea service
Apr 25 03:38:23 raspberrypi nymea-networkmanager[4875]:  I | Application:
Shutting down bluetooth service
Apr 25 03:38:23 raspberrypi nymea-networkmanager[4875]:  I |
BluetoothServer: Destroy bluetooth server.
Apr 25 03:38:23 raspberrypi nymea-networkmanager[4875]:  I | Application:
Shutting down network-manager service
Apr 25 03:38:23 raspberrypi nymea-networkmanager[4875]:  I |
NetworkManager: Destroy network manager.
Apr 25 03:38:23 raspberrypi systemd[1]: Stopping Daemon for nymea to
configure wifi network using a Bluetooth LE connection...
Apr 25 03:38:23 raspberrypi systemd[1]: Stopped Daemon for nymea to
configure wifi network using a Bluetooth LE connection.
Apr 25 03:38:23 raspberrypi systemd[1]: Started Daemon for nymea to
configure wifi network using a Bluetooth LE connection.
Apr 25 03:38:23 raspberrypi nymea-networkmanager[4888]:  I | Application:
Using configuration file from: "/etc/nymea/nymea-networkmanager.conf"
Apr 25 03:38:23 raspberrypi nymea-networkmanager[4888]:  I | Application:
=====================================
Apr 25 03:38:23 raspberrypi nymea-networkmanager[4888]:  I | Application:
Starting nymea-networkmanager "0.3.0"
Apr 25 03:38:23 raspberrypi nymea-networkmanager[4888]:  I | Application:
=====================================
Apr 25 03:38:23 raspberrypi nymea-networkmanager[4888]:  I | Application:
Advertising name: "Intelligent Demographics"
Apr 25 03:38:23 raspberrypi nymea-networkmanager[4888]:  I | Application:
Platform name: "nymea-box"
Apr 25 03:38:23 raspberrypi nymea-networkmanager[4888]:  I | Application:
Mode: Core::Mode(ModeOffline)
Apr 25 03:38:23 raspberrypi nymea-networkmanager[4888]:  I | Application:
Timeout: 60
Apr 25 03:38:23 raspberrypi nymea-networkmanager[4888]:  W | NymeaService:
Invalid D-Bus HardwareManager interface.
Apr 25 03:38:23 raspberrypi nymea-networkmanager[4888]:  W | NymeaService:
Could not init nymea D-Bus services
Apr 25 03:38:23 raspberrypi nymea-networkmanager[4888]:  I |
NetworkManager: Start the network manager.
Apr 25 03:38:23 raspberrypi nymea-networkmanager[4888]:  I |
NetworkManager: Initialize network manager
Apr 25 03:38:23 raspberrypi nymea-networkmanager[4888]:  I |
NetworkManager: Version: "1.6.2"
Apr 25 03:38:23 raspberrypi nymea-networkmanager[4888]:  I |
NetworkManager: State changed: "NetworkManagerStateConnectedGlobal"
Apr 25 03:38:23 raspberrypi nymea-networkmanager[4888]:  I | Application:
NetworkManager::NetworkManagerState(NetworkManagerStateConnectedGlobal)
Apr 25 03:38:23 raspberrypi nymea-networkmanager[4888]:  I |
NetworkManager: Connectivity state changed: "Full"
Apr 25 03:38:23 raspberrypi nymea-networkmanager[4888]:  I |
NetworkManager: Networking enabled
Apr 25 03:38:23 raspberrypi nymea-networkmanager[4888]:  I | Application:
Networkmanager networking is now enabled
Apr 25 03:38:23 raspberrypi nymea-networkmanager[4888]:  I |
NetworkManager: Wireless networking enabled
Apr 25 03:38:23 raspberrypi nymea-networkmanager[4888]:  I | Application:
Networkmanager wireless networking is now enabled
Apr 25 03:38:23 raspberrypi nymea-networkmanager[4888]:  I |
NetworkManager: [+] NetworkDevice("lo" - "Generic",
"NetworkDeviceStateUnmanaged")
Apr 25 03:38:24 raspberrypi nymea-networkmanager[4888]:  I |
NetworkManager: [+] WirelessNetworkDevice("wlan0", "B8:27:EB:5B:5D:FA",
WirelessNetworkDevice::Mode(ModeInfrastructure), 24 [Mb/s],
"NetworkDeviceStateActivated")
Apr 25 03:38:24 raspberrypi nymea-networkmanager[4888]:  I |
NetworkManager: [+] WiredNetworkDevice("usb0", "3A:25:2C:17:D9:31", 0
[Mb/s], false, "NetworkDeviceStateUnmanaged")
Apr 25 03:38:24 raspberrypi nymea-networkmanager[4888]:  I |
NetworkManager: Settings: [+] NetworkConnection("TP-LINK_F9993C",
"{5197441b-a3d4-41e7-81e9-f02a161bd5de}", "", "802-11-wireless",
"22.02.2019 03:07")
Apr 25 03:38:24 raspberrypi nymea-networkmanager[4888]:  I |
NetworkManager: Settings: [+] NetworkConnection("TP-LINK7",
"{f857043a-381f-4a5c-bad5-3cf2f9d191c5}", "", "802-11-wireless",
"25.04.2019 03:36")
Apr 25 03:38:24 raspberrypi nymea-networkmanager[4888]:  I |
NetworkManager: Settings: [+] NetworkConnection("upstairs 3",
"{d6df8331-4311-4e1f-9353-68ed0ee1c7e1}", "", "802-11-wireless",
"08.02.2019 19:39")
Apr 25 03:38:24 raspberrypi nymea-networkmanager[4888]:  I |
NetworkManager: Settings: [+] NetworkConnection("TP-LINK2",
"{618c63ee-3abf-4cee-8945-eb09b6d9681c}", "", "802-11-wireless",
"22.02.2019 17:26")
Apr 25 03:38:24 raspberrypi nymea-networkmanager[4888]:  I |
NetworkManager: Settings: [+] NetworkConnection("TP-LINK3",
"{1aaff37a-3518-4e86-8d5a-c63e766af23f}", "", "802-11-wireless",
"22.02.2019 19:20")
Apr 25 03:38:24 raspberrypi nymea-networkmanager[4888]:  I |
NetworkManager: Settings: [+] NetworkConnection("TP-LINK4",
"{6faba528-c912-44d0-a755-527bf91698e3}", "", "802-11-wireless",
"05.03.2019 18:29")
Apr 25 03:38:24 raspberrypi nymea-networkmanager[4888]:  I |
NetworkManager: The network manager is now available
Apr 25 03:38:24 raspberrypi nymea-networkmanager[4888]:  I | Application:
Init is running
Apr 25 03:38:24 raspberrypi nymea-networkmanager[4888]:  I |
NetworkManager: Network manager initialized successfully.
Apr 25 03:38:29 raspberrypi nymea-networkmanager[4888]:  I | Application:
Post run service

rnehrboss, 2019-04-25:
An interesting thing.. If I choose the device in the client app, it goes
forward to connect, but immediately back to the Devices screen and it's no
longer showing in the list:

In this case it is advertising bluetooth on the client app:

-- Logs begin at Thu 2019-04-25 03:51:26 UTC. --
Apr 25 03:55:05 raspberrypi nymea-networkmanager[406]:  I | Application:
Shutting down bluetooth service
Apr 25 03:55:05 raspberrypi nymea-networkmanager[406]:  I |
BluetoothServer: Destroy bluetooth server.
Apr 25 03:55:05 raspberrypi nymea-networkmanager[406]:  I |
BluetoothServer: Controller state disonnected.
Apr 25 03:55:05 raspberrypi nymea-networkmanager[406]:  I |
BluetoothServer: Delete network service
Apr 25 03:55:05 raspberrypi nymea-networkmanager[406]:  I |
BluetoothServer: Delete wireless service
Apr 25 03:55:05 raspberrypi nymea-networkmanager[406]:  I | Application:
Shutting down network-manager service
Apr 25 03:55:05 raspberrypi nymea-networkmanager[406]:  I | NetworkManager:
Destroy network manager.
Apr 25 03:55:05 raspberrypi systemd[1]: Stopping Daemon for nymea to
configure wifi network using a Bluetooth LE connection...
Apr 25 03:55:06 raspberrypi systemd[1]: Stopped Daemon for nymea to
configure wifi network using a Bluetooth LE connection.
Apr 25 03:55:06 raspberrypi systemd[1]: Started Daemon for nymea to
configure wifi network using a Bluetooth LE connection.
Apr 25 03:55:06 raspberrypi nymea-networkmanager[902]:  I | Application:
Using configuration file from: "/etc/nymea/nymea-networkmanager.conf"
Apr 25 03:55:06 raspberrypi nymea-networkmanager[902]:  I | Application:
=====================================
Apr 25 03:55:06 raspberrypi nymea-networkmanager[902]:  I | Application:
Starting nymea-networkmanager "0.3.0"
Apr 25 03:55:06 raspberrypi nymea-networkmanager[902]:  I | Application:
=====================================
Apr 25 03:55:06 raspberrypi nymea-networkmanager[902]:  I | Application:
Advertising name: "Intelligent Demographics"
Apr 25 03:55:06 raspberrypi nymea-networkmanager[902]:  I | Application:
Platform name: "nymea-box"
Apr 25 03:55:06 raspberrypi nymea-networkmanager[902]:  I | Application:
Mode: Core::Mode(ModeOffline)
Apr 25 03:55:06 raspberrypi nymea-networkmanager[902]:  I | Application:
Timeout: 60
Apr 25 03:55:06 raspberrypi nymea-networkmanager[902]:  W | NymeaService:
Invalid D-Bus HardwareManager interface.
Apr 25 03:55:06 raspberrypi nymea-networkmanager[902]:  W | NymeaService:
Could not init nymea D-Bus services
Apr 25 03:55:06 raspberrypi nymea-networkmanager[902]:  I | NetworkManager:
Start the network manager.
Apr 25 03:55:06 raspberrypi nymea-networkmanager[902]:  I | NetworkManager:
Initialize network manager
Apr 25 03:55:06 raspberrypi nymea-networkmanager[902]:  I | NetworkManager:
Version: "1.6.2"
Apr 25 03:55:06 raspberrypi nymea-networkmanager[902]:  I | NetworkManager:
State changed: "NetworkManagerStateConnectedGlobal"
Apr 25 03:55:06 raspberrypi nymea-networkmanager[902]:  I | Application:
NetworkManager::NetworkManagerState(NetworkManagerStateConnectedGlobal)
Apr 25 03:55:06 raspberrypi nymea-networkmanager[902]:  I | NetworkManager:
Connectivity state changed: "Full"
Apr 25 03:55:06 raspberrypi nymea-networkmanager[902]:  I | NetworkManager:
Networking enabled
Apr 25 03:55:06 raspberrypi nymea-networkmanager[902]:  I | Application:
Networkmanager networking is now enabled
Apr 25 03:55:06 raspberrypi nymea-networkmanager[902]:  I | NetworkManager:
Wireless networking enabled
Apr 25 03:55:06 raspberrypi nymea-networkmanager[902]:  I | Application:
Networkmanager wireless networking is now enabled
Apr 25 03:55:07 raspberrypi nymea-networkmanager[902]:  I | NetworkManager:
[+] NetworkDevice("lo" - "Generic", "NetworkDeviceStateUnmanaged")
Apr 25 03:55:07 raspberrypi nymea-networkmanager[902]:  I | NetworkManager:
[... 21 lines cut ...]
"22.02.2019 19:20")
Apr 25 03:55:08 raspberrypi nymea-networkmanager[902]:  I | NetworkManager:
Settings: [+] NetworkConnection("TP-LINK_F9993C",
"{5197441b-a3d4-41e7-81e9-f02a161bd5de}", "", "802-11-wireless",
"22.02.2019 03:07")
Apr 25 03:55:08 raspberrypi nymea-networkmanager[902]:  I | NetworkManager:
Settings: [+] NetworkConnection("upstairs 3",
"{d6df8331-4311-4e1f-9353-68ed0ee1c7e1}", "", "802-11-wireless",
"08.02.2019 19:39")
Apr 25 03:55:08 raspberrypi nymea-networkmanager[902]:  I | NetworkManager:
Settings: [+] NetworkConnection("TP-LINK4",
"{1fca9cb8-99e3-4d13-95d3-f90910ebb1be}", "", "802-11-wireless",
"08.03.2019 02:33")
Apr 25 03:55:08 raspberrypi nymea-networkmanager[902]:  I | NetworkManager:
Settings: [+] NetworkConnection("TP-LINK5",
"{4d8ae010-be6d-4e08-870d-d4e81cb98ab5}", "", "802-11-wireless",
"09.03.2019 02:48")
Apr 25 03:55:08 raspberrypi nymea-networkmanager[902]:  I | NetworkManager:
The network manager is now available
Apr 25 03:55:08 raspberrypi nymea-networkmanager[902]:  I | Application:
Init is running
Apr 25 03:55:08 raspberrypi nymea-networkmanager[902]:  I | NetworkManager:
Network manager initialized successfully.
Apr 25 03:55:13 raspberrypi nymea-networkmanager[902]:  I | Application:
Post run service

t-mon, 2019-04-25:
I still don't see any logs regarding the Bluetooth server :) 
they should come after the network manager initialization and show why the Bluetooth server is starting and if the Bluetooth server is starting. If those are the full logs, the server is not advertising, but maybe the Bluetooth host itself, but no server behind it.

rnehrboss, 2019-04-26:
It seems that after reboot, it's still connected to wifi AP, but advertises
on bluetooth.  Here is a log after reboot.
It's like after reboot it doesn't sense that it is connected but it is.

-- Logs begin at Fri 2019-04-26 17:17:01 UTC. --
Apr 26 18:14:31 raspberrypi nymea-networkmanager[466]:  I |
BluetoothServer: Create NetworkService.
Apr 26 18:14:31 raspberrypi nymea-networkmanager[466]:  I |
BluetoothServer: NetworkService: Notify state changed "\x07"
Apr 26 18:14:31 raspberrypi nymea-networkmanager[466]:  I |
BluetoothServer: NetworkService: Notify networking enabled changed: enabled
Apr 26 18:14:31 raspberrypi nymea-networkmanager[466]:  I |
BluetoothServer: NetworkService: Notify wireless networking enabled
changed: enabled
Apr 26 18:14:31 raspberrypi nymea-networkmanager[466]:  I |
BluetoothServer: Create WirelessService.
Apr 26 18:14:32 raspberrypi nymea-networkmanager[466]:  I |
BluetoothServer: Start advertising "Intelligent Demographics"
"B8:27:EB:67:B9:16"
Apr 26 18:14:32 raspberrypi nymea-networkmanager[466]:  I |
BluetoothServer: Controller state advertising...
Apr 26 18:14:32 raspberrypi nymea-networkmanager[466]:  I | Application:
Bluetooth server started
Apr 26 18:14:32 raspberrypi nymea-networkmanager[466]:  I |
BluetoothServer: Bluetooth host in discoverable mode.
Apr 26 18:21:13 raspberrypi nymea-networkmanager[466]:  I |
BluetoothServer: Bluetooth host in connectable mode.

If I use the berrylan client again,  it let's me connect to the same AP,
and then goes away

rnehrboss, 2019-09-16:
Any further thoughts on this?

t-mon, 2019-09-16:
I'm currently restructuring the whole project, which should make things much easier, and hopefully also fix this issue. I'll keep this in mind when testing. Expect a new version in the near future. :)

rnehrboss, 2019-09-16:
Thanks Simon.  Do you ever encounter situations where the login doesn't
"stick"?

t-mon, 2019-09-16:
Not sure what you mean with stick? You mean the network does not reconnect automatically or stays connected?

rnehrboss, 2019-09-16:
I think I've solved the rebroadcasting issue.  Looks like we had some
legacy code that was also trying to using BT stack. :(

Hey Simon, where is the device name that is showing up in the app being
set?

Thanks again!
Richard

mzanetti, 2019-09-20:
@rnehrboss Should be in `/etc/nymea/nymea-networkmanager.conf`

mzanetti, 2019-11-07:
The updates @t-mon referred to have been released and should have improved this behavior.

closed 2019-11-07 by mzanetti