nymea/berrylan #19: support to raspian buster

richyxi, 2019-08-13:
does it work with the new raspian OS?

Boernsman, 2019-08-14:
Out plan is to support Buster within the next weeks, we don't have any packages for it yet. Probably BerryLan will work when you compile and install the nymea-networkmanager yourself.

richyxi, 2019-08-14:
Yeah i was thinking the same by doing it manually, ill report any result if
that helps to you.

Very awsome proyect btw!!!

brandonhilkert, 2019-08-14:
FWIW - We followed the `README` on latest buster lite and it worked great!

Boernsman, 2019-08-14:
Nice, thanks!

mzanetti, 2019-11-07:
FWIW, we have added a repository for debian buster now too:

"deb http://repository.nymea.io buster main"

However, please note that this does not support the Pi0 at this point as it's built with a regular armhf compiler instead of Raspberry Pi's modded armel compiler. It should work fine on RaspberryPi 2, 3 and 4.

closed 2019-11-07 by mzanetti