nymea/berrylan #20: Is there any way to change the network the pi is connected to?

HarrisonFretwell, 2019-08-13:
I want to be able to change the selected network after having already connected to one, effectively wiping the installation and starting again. Is there anyway of doing this? I presume the wifi credentials are stored in a file somewhere so is it as simple as resetting/deleting that file?

Boernsman, 2019-08-13:
You can change this behavior in the [nymea-networkmanager](https://github.com/guh/nymea-networkmanager) - default behavior is offline -> takes the BT down a soon as the device is online, you might want to change it to "start" ->  BT server is up for 3min after boot.

HarrisonFretwell, 2019-08-14:
It actually looks like that behaviour is enabled by default for my device? This is using the latest release of BerryLan (The version that comes packaged with stretch). Regardless that fixes my issue.

closed 2019-08-14 by HarrisonFretwell