nymea/berrylan #21: Switching Wifi connections when original becomes unavailable

brandonhilkert, 2019-08-14:
First off, thank you so much for this project!

Second, just a quick question (sorry if this should go somewhere else). Let's say we set up a pi with BerryLAN and all goes well. Sometime down the road the wifi connection we connected to is no longer available (e.g. moved locations, new router, etc.), will the box look accessible in the app again? or would we have to reset something on the pi for it to broadcast over BLE again?

Boernsman, 2019-08-14:
It depends. The nymea-networkmanager is by default confiured in the "offline" mode, means the (BerryLan) Bluetooth servers starts as soon as the device gets offline. I guess this is what you are searching for, more about the nymea-networkmanager modes here: https://github.com/guh/nymea-networkmanager

brandonhilkert, 2019-08-14:
Ah, yup. I later stumbled on that and those options indeed answer the question. Sorry for the putting that in the wrong place 🎉

closed 2019-08-14 by brandonhilkert