nymea/berrylan #22: wrong password - not

shilbert01, 2019-08-15:
Berrylan on Android  detects the Pi, lists the local WLAN and asks for the password. Even though the password is entered correctly, it is rejected as being wrong.

On a side note it detects only on Wifi network despite 3-4 being nearby.

Is there a log file on the Pi to debug this ?

Boernsman, 2019-08-16:
The nymea-network-manager has a debug option ` nymea-network-manager  --debug`. With journalctl you should be able to get the debug output `journalctl -f -u  nymea-network-manager `

shilbert01, 2019-08-18:
I am seeing two issues. 

1) scanning for access point seems to be unreliable. It does not consistently detect my own access point(s). It is of note that I did not make use of the berrylan image but installed on top of a Raspian. When scanning for APs it will sometimes detect my AP and sometime it won't.

Wifi was working without issues on stock Raspian before installation of berrylan.

If will try with the supplied image to rule out issues introduced by installing manually.

shilbert01, 2019-08-18:
Another strange thing. Sometimes the Android app will detect the Pi as "BT WLAN setup" and sometimes it will display the hostname (e.g. smarmirror-mini2)

shilbert01, 2019-08-18:
sudo iwlist wlan0 scan

yields no results or sometimes just one far away access point. So this whole issue seem to be related to the Pi3B+ Wifi rather then berrylan/nymea-networkmanager

shilbert01, 2019-08-18:
After cleaning up wpa_supplicant it now works.

mzanetti, 2019-11-07:
Closing this, assuming a configuration mistake of the raspberry pi

closed 2019-11-07 by mzanetti