nymea/berrylan #23: wifi network not detected - solved

shilbert01, 2019-08-20:
I downloaded and burned your image for the Pi. The Android app would not detect my local Wifi.

Turns out that Raspian/Pi seems to have trouble with Wifi that is on channel 12. Switching to another Wifi channel fixed the issue. 

I believe channel 12 and 13 are not available in all countries.

Boernsman, 2019-08-20:
That's correct, for example in the EU are the WiFi channels 12 and 13 not available. This must be improved. Thanks for the report.

mzanetti, 2020-02-24:
@t-mon Can we fix something in nymea-networkmanager for this?

mzanetti, 2020-11-05:
I've looked into this now and it turns out that berrylan (and nymea-networkmanager) support this just fine. However, there seems to be an issue with some Raspberry Pi models. In my tests it seemed to work fine with some of them, but not with others.

According to some other discussions in the internet that seems to be the fault of the raspberrys WiFi firmware and it should be fixable by upgrading it.

```
sudo apt-get update
sudo apt-get full-upgrade
sudo rpi-update
```
Also, make sure you haven't set the WiFi country code to US as it's not allowed to use channel 12 and 13 there. See `iw reg get` to get the current region and use `sudo iw reg set <COUNTRY>` selecting the appropriate country code from this list: https://en.wikipedia.org/wiki/ISO_3166-1_alpha-2#Officially_assigned_code_elements

Closing this issue now. If you still have issues after this, please report back.

closed 2020-11-05 by mzanetti