nymea/berrylan #25: install without predefined image

rudiratlos, 2019-09-04:
Hi,
due to my own project (own image), I can not use your image. How can I install berrylan on the newest buster light OS? I use rpi3B+ and 4.

mzanetti, 2019-09-04:
Packages for buster will be uploaded very soon.
We've already got reports that the stretch version works on buster too. For testing now you can try to install that one.

`sudo apt-add-repository http://repository.nymea.io stretch main`
`sudo apt-get install nymea-networkmanager`

rudiratlos, 2019-09-04:
I've already installed this on a plain Buster Raspbian, but it didn't work.

t-mon, 2019-09-05:
Hi, i just released a fixed version of the nymea-networkmanager, and it should also be available for debian buster now. Please let us know if it worked for you!

mzanetti, 2020-01-18:
Should be fixed now

closed 2020-01-18 by mzanetti