nymea/berrylan #26: not working in kali

connected201, 2019-10-11:
hi, in kali os, i cannot discovery my raspberry :(
how i can fix it?
hardware: raspberry pi 3
OS : https://images.offensive-security.com/arm-images/kali-linux-2019.3a-rpi3-nexmon.img.xz
best regards

mzanetti, 2019-10-11:
Did you install nymea-networkmanager on the Raspberry Pi3?

As far as I know, Kali ist based on Debian, so our repository should work. A quick google search for it gives this link: https://docs.kali.org/general-use/kali-linux-sources-list-repositories

You'd need to add the nymea repository and then install nymea-networkmanager.
The issue though is that we don't build packages specifically for kali. You'll have to try, probably using the debian buster repo could work, however, it could also break something on your kali image. You could also try to install libnymea-networknamanager and nymea-networkmanager packages manually by downloading them from the repository and using `sudo dpkg -i <packagename>` to install them...

Our repo is at `deb http://repository.nymea.io buster main`. Good luck.

mzanetti, 2019-10-11:
As Kali is not a supported platform, I'm closing this bug. You are however most welcome to continue discussing this in the forum at https://forum.nymea.io/c/berrylan

closed 2019-10-11 by mzanetti

mzanetti, 2019-10-11:
Is it now working for you or still a problem?

About `systemctl disable dhcpcd` not working: If it says `service not found` it is ok, most likely kali does not use dhcpcd in the first place. Plain debian/ubuntu don't have that either, it's just Raspbian which uses that for some reason.