nymea/berrylan #30: Install on RPi 4 and Android app fails to find WiFi ap

bafu, 2019-11-19:
Hi BeryLan team, thanks for the awesome work!

I install BerryLan on RPi 4 by the steps below, and the Android app can not find any WiFi ap to connect (keep showing a spinner on screen):

1. Setup repository

    ```
    pi@raspberrypi:~ $ sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-key A1A19ED6
    Executing: /tmp/apt-key-gpghome.DabmpFMYwt/gpg.1.sh --keyserver keyserver.ubuntu.com --recv-key A1A19ED6
    gpg: key 457A6EE4A1A19ED6: public key "Simon Stuerz <simon.stuerz@guh.guru>" imported
    gpg: Total number processed: 1
    gpg:               imported: 1

    pi@raspberrypi:~ $ cat /etc/apt/sources.list.d/nymea.list
    deb http://repository.nymea.io buster main raspberry
    ```

1. Install nymea-networkmanager

    ```
    pi@raspberrypi:~ $ sudo apt install nymea-network-manager
    ```

1. Disable dhcpcd

    ```
    pi@raspberrypi:~ $ sudo systemctl disable dhcpcd
    ```

---
I'm sure that RPi 4 did not connect to any WiFi ap (issue #7), and my ap is on channel #1 (issue #23)

If you can share the debugging steps with me, I'm happy to share the debugging log messages.

Thanks!

t-mon, 2019-11-19:
Hi @bafu 

thanks for the feedback, you might have found an issue and I'll have a closer look into that. I'll let you know once I found the issue, I might have an idea whats going wrong.

mzanetti, 2020-01-18:
Is this still valid or has it been fixed with the last update?

mzanetti, 2020-03-05:
This should be fixed now

closed 2020-03-05 by mzanetti