nymea/berrylan #31: nymea on bananapi (ubuntu server)

legnaro, 2019-11-26:
Hi,

I'm big fan of this project and I have been used berrylan with success in raspberry zero (kali linux). Now I trying to use a new hardware "BananaPi M2 zero" with UbuntuServer, but I've got a message:

pi@bpi-iot-ros-ai:/home$ sudo nymea-networkmanager --mode always -d
nymea-networkmanager: /usr/lib/arm-linux-gnueabihf/libQt5DBus.so.5: version Qt_5' not found (required by nymea-networkmanager) nymea-networkmanager: /usr/lib/arm-linux-gnueabihf/libQt5Core.so.5: version Qt_5' not found (required by nymea-networkmanager)
nymea-networkmanager: /usr/lib/arm-linux-gnueabihf/libQt5DBus.so.5: version Qt_5' not found (required by /usr/lib/arm-linux-gnueabihf/libnymea-networkmanager.so.1) nymea-networkmanager: /usr/lib/arm-linux-gnueabihf/libQt5Core.so.5: version Qt_5' not found (required by /usr/lib/arm-linux-gnueabihf/libnymea-networkmanager.so.1)
nymea-networkmanager: /usr/lib/arm-linux-gnueabihf/libQt5Core.so.5: version `Qt_5' not found (required by /usr/lib/arm-linux-gnueabihf/libnymea-gpio.so.1)
pi@bpi-iot-ros-ai:/home$

If I try for another way like compile the project, th system return this message:
Project ERROR: nymea-networkmanager development package not found

What I tried?
-To install:  libnymea-networkmanager-dev, libnymea-gpio-dev, qt5-default qtbase5-dev qtbase5-dev-tools libqt5bluetooth5 qtconnectivity5-dev libnymea-networkmanager-dev libnymea-gpio-dev git

- Compile project following the steps of wiki: 
   sudo nano /etc/apt/sources.list.d/nymea.list
   deb http://repository.nymea.io xenial main
   deb-src http://repository.nymea.io xenial main
   ...
   (same error of package)

Apparently is posible to run nymea-networkmanager in this hardware, but I don't know how the best way to fix these issue.

tks

mzanetti, 2019-11-26:
please post the output of
`lsb_release -a`
and
`apt-cache policy libqt5core5a`
and
`file /usr/lib/arm-linux-gnueabihf/libQt5Core.so.5*`

legnaro, 2019-11-26:
hi mzanetti, thank you for your answer!

Follow the Output:

pi@bpi-iot-ros-ai:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.6 LTS
Release:	16.04
Codename:	xenial

pi@bpi-iot-ros-ai:~$ apt-cache policy libqt5core5a
libqt5core5a:
  Installed: 5.5.1+dfsg-16ubuntu7.6
  Candidate: 5.5.1+dfsg-16ubuntu7.6
  Version table:
 *** 5.5.1+dfsg-16ubuntu7.6 500
        500 http://ports.ubuntu.com xenial-security/main armhf Packages
        500 http://ports.ubuntu.com xenial-updates/main armhf Packages
        100 /var/lib/dpkg/status
     5.5.1+dfsg-16ubuntu7 500
        500 http://ports.ubuntu.com xenial/main armhf Packages

pi@bpi-iot-ros-ai:~$ file /usr/lib/arm-linux-gnueabihf/libQt5Core.so.5*
/usr/lib/arm-linux-gnueabihf/libQt5Core.so.5:     symbolic link to libQt5Core.so.5.5.1
/usr/lib/arm-linux-gnueabihf/libQt5Core.so.5.5:   symbolic link to libQt5Core.so.5.5.1
/usr/lib/arm-linux-gnueabihf/libQt5Core.so.5.5.1: ELF 32-bit LSB shared object, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib, BuildID[sha1]=c29ab4326bd0ba3a208643349065b34e7cee4152, stripped

t-mon, 2019-11-27:
Hi @legnaro,
I'm afraif with this Qt Version the project will not work. We are using the QBluetooth libraries for creating the bluetooth server, and also the bluez needs to support that feature. QBluetooth LE server support arrived with Qt Version 5.7 and in bluez it left the experimental state with bluez version 5.46 (if I remember correctly). This information is definitely missing in our README. 
What you could try is back-porting a Qt Version > 5.7 to your platform (quiet a task) and check the bluez version.

mzanetti, 2019-11-27:
Isn't there a newer Ubuntu version available for the banana pi? Xenial is nearly 4 years old now... Ubuntu Buster 18.04  would make this a lot easier...

legnaro, 2019-11-27:
Excelent! very good tips! First, I´ll try to check if its posible to update the Qt (I googled and saw somebody trying do it) and then I´ll try to update the SO to 19.10 Eoan Ermine, apparently it was avaliable for testing in bananapi m2 zero (https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v2.4)

tks

legnaro, 2019-12-05:
Hi folks, some news!

I installed ubuntu 19.1o Eoan Ermine, the situation is better, now I can start the bluetooth by nymea, but the system stops in the step 2, doesn ask for password, follow the log:

ubuntu@bpi-m2z:~$ sudo nymea-networkmanager --mode always -d
 I | Application: Using configuration file from: "/etc/nymea/nymea-networkmanager.conf"
 I | Application: =====================================
 I | Application: Starting nymea-networkmanager "0.5.0+201910170244~bionic1"
 I | Application: =====================================
 I | Application: Advertising name: "BT WLAN setup"
 I | Application: Platform name: "nymea-box"
 I | Application: Mode: Core::ModeAlways
 I | Application: Timeout: 60
 W | NymeaService: Invalid D-Bus HardwareManager interface.
 W | NymeaService: Could not init nymea D-Bus services
 I | NetworkManager: Start the network manager.
 I | NetworkManager: Initialize network manager
 I | NetworkManager: Version: "1.20.4"
 I | NetworkManager: State changed: "NetworkManagerStateConnectedGlobal"
 I | NetworkManager: Connectivity state changed: "None"
 I | NetworkManager: Networking enabled
 I | NetworkManager: Wireless networking enabled
 I | NetworkManager: [+] NetworkDevice("lo" - "Generic", "NetworkDeviceStateUnmanaged")
 I | NetworkManager: [+] WiredNetworkDevice("eth0", "02:42:F3:AF:10:B1", 0 [Mb/s], false, "NetworkDeviceStateUnmanaged") 
 I | NetworkManager: [+] WirelessNetworkDevice("wlan0", "CC:B8:A8:AA:78:FA", WirelessNetworkDevice::ModeInfrastructure, 0 [Mb/s], "NetworkDeviceStateUnavailable") 
 I | NetworkManager: The network manager is now available
 I | Application: Init is running...
 I | NetworkManager: Network manager initialized successfully.
 I | Application: Post run service
 I | Application: Start the bluetooth service because of "always" mode.
 I | Application: Start the service...
 W | NymeaService: Could not enable/disable bluetooth hardware resource. D-Bus interface not available.
 I | NetworkManagerBluetoothServer: -------------------------------------
 I | NetworkManagerBluetoothServer: Starting bluetooth server...
 I | NetworkManagerBluetoothServer: -------------------------------------
 I | NetworkManagerBluetoothServer: Local device "bpi-m2z" "CC:B8:A8:AA:78:FB"
 W | qt.bluetooth: Using BlueZ kernel ATT interface
 I | NetworkManagerBluetoothServer: Serial number not specified. Using system uuid from /etc/machine-id as serialnumber.
 I | NetworkManagerBluetoothServer: Create NetworkService.
 I | NetworkManagerBluetoothServer: Create WirelessService.
 I | NetworkManagerBluetoothServer: WirelessService: Using WirelessNetworkDevice("wlan0", "CC:B8:A8:AA:78:FA", WirelessNetworkDevice::ModeInfrastructure, 0 [Mb/s], "NetworkDeviceStateUnavailable") 
 I | NetworkManagerBluetoothServer: Start advertising "BT WLAN setup" "CC:B8:A8:AA:78:FB"
 I | NetworkManagerBluetoothServer: Controller state advertising...
 I | Application: Bluetooth server started
 I | NetworkManagerBluetoothServer: Bluetooth host in discoverable mode.
 I | NetworkManagerBluetoothServer: Device connected "7B:05:64:E8:0C:2E"
 I | NetworkManagerBluetoothServer: Controller state connected. "7B-05-64-E8-0C-2E" "7B:05:64:E8:0C:2E"
 I | Application: Bluetooth client connected
 I | NetworkManagerBluetoothServer: Client connected "7B-05-64-E8-0C-2E" "7B:05:64:E8:0C:2E"
 W | qt.bluetooth.bluez: sending error response; request: 16 handle: 51 code: 10
 W | qt.bluetooth.bluez: sending error response; request: 8 handle: 1 code: 10
 W | qt.bluetooth.bluez: sending error response; request: 8 handle: 13 code: 10
 W | qt.bluetooth.bluez: sending error response; request: 8 handle: 14 code: 10
 W | qt.bluetooth.bluez: sending error response; request: 8 handle: 22 code: 10
 W | qt.bluetooth.bluez: sending error response; request: 8 handle: 23 code: 10
 W | qt.bluetooth.bluez: sending error response; request: 8 handle: 25 code: 10
 W | qt.bluetooth.bluez: sending error response; request: 8 handle: 26 code: 10
 W | qt.bluetooth.bluez: sending error response; request: 8 handle: 38 code: 10
 W | qt.bluetooth.bluez: sending error response; request: 8 handle: 39 code: 10
 W | qt.bluetooth.bluez: sending error response; request: 8 handle: 49 code: 10
 I | NetworkManagerBluetoothServer: NetworkService: Descriptor written "{00002902-0000-1000-8000-00805f9b34fb}" "\x01\x00"
 I | NetworkManagerBluetoothServer: WirelessService: Descriptor written "{00002902-0000-1000-8000-00805f9b34fb}" "\x01\x00"
 I | NetworkManagerBluetoothServer: WirelessService: Descriptor written "{00002902-0000-1000-8000-00805f9b34fb}" "\x01\x00"
 I | NetworkManagerBluetoothServer: Got command stream "{\n    \"c\": 0\n}\n"
 I | NetworkManagerBluetoothServer: WirelessService: Start streaming response data: 21 bytes
 I | NetworkManagerBluetoothServer: WirelessService: Finished streaming response data
 I | NetworkManagerBluetoothServer: Got command stream "{\n    \"c\": 5\n}\n"
 I | NetworkManagerBluetoothServer: There is currently no access active accesspoint
 I | NetworkManagerBluetoothServer: WirelessService: Start streaming response data: 53 bytes
 I | NetworkManagerBluetoothServer: WirelessService: Finished streaming response data

tks

ericmigi, 2020-01-12:
@legnaro did you get berrylan to work on eoan?

mzanetti, 2020-01-13:
FWIW, there is now a repository for eoan.

`deb http://repository.nymea.io eoan main`

legnaro, 2020-01-14:
Hi Eric/mzanetti, In the previous version I got error in the password step 2 (never ask for the password) I going to try with this new repository.

tks a lot!

mzanetti, 2020-01-18:
Closing this now. We're supporting armhf and arm64 builds for Ubuntu bionic, disco and eoan, Debian armhf and arm64 builds for stretch and buster as well as raspbian builds (armv6) for stretch and buster.

That's probably as far as we can go with the resources we have at this point and I believe some of them should work on a banana pi too.

closed 2020-01-18 by mzanetti