nymea/berrylan #33: Supported hardware for the app

wilkgr76, 2019-12-29:
The Play Store reports that only my ZTE Blade V6 is compatible with the app:
[image]
For reference, this is a budget phone from 2014 that runs Android 5.0. I do not know where the phone is, and would like to use Berrylan on my actual phone - a Moto G6. 

Well, no problem, I thought, I'll just grab the apk and install that. No bueno:
[image]

The Moto G6 has Bluetooth 4.2, so I'm not sure why this is marked as 'not compatible'.

wilkgr76, 2019-12-29:
Attempting to install via adb gives an error message that might be more useful:
```
$ adb install berrylan.apk 
adb: failed to install berrylan.apk: Failure [INSTALL_FAILED_NO_MATCHING_ABIS: Failed to extract native libraries, res=-113]
```

mzanetti, 2020-01-02:
Ok, after some investigation I believe this is because your phone only supports 32 bit builds... Will see how to create a multi-arch package

mzanetti, 2020-01-08:
Can you try this package on your device please: https://downloads.nymea.io/tmp/berrylan-android-armhf.apk

It's a bit odd... google doesn't allow uploading 32bit packages any more to their store yet. Only 64bir or multiarch packages are allowed. I still need to figure out how to create the multiarch packages.

mzanetti, 2020-01-11:
Ok. I've done some more research on the topic. While apparently it is possible to upload multiple architectures of .apk files, it is pretty complicated to do so as the play store has a lot of requirements for that (e.g. version codes in certain format/order, upload order and more).
Now, this also is discouraged by google as the new way is to upload .aab multi-arch app bundles. This is supported by Qt as of 5.14.0 and I've git it mostly working here already. Odd thing is that there is a bug in Qt 5.14.0 which breaks styling on Android so we have to wait for at least 5.14.1 until we can move to the .aab format. So expecting the styling bug to be fixed in Qt within the next month, I'd not spend the effort to do the deprecated way of uploading multiple apks. 

For now I'd recommend to use the above 32 bit .apk for those devices and I'll move to .aab to support all devices as soon as the styling bug is fixed in Qt.

For the reference, this is the Qt bug I'm talking about: https://bugreports.qt.io/browse/QTBUG-81216

wilkgr76, 2020-01-19:
The 32-bit version installed successfully. Thank you for your help!

mzanetti, 2020-04-03:
Hey everyone, finally all the required fixes are in place and the latest beta channel build should now support all android devices with android 5.0 or greater.

I'd appreciate if you guys could join the beta program and test the version 1.0.35 in there.

There should be an option to join the beta program for berrylan in the play store by default, but in case it won't show up for someone, you can use this link to join the beta program:
https://play.google.com/apps/testing/io.guh.berrylan

Please let me know if it works for you on those devices now so I can release the package to stable.

mzanetti, 2020-04-04:
Ok. This has now been fixed and released to the play store.

Thanks again for reporting.

closed 2020-04-04 by mzanetti