nymea/berrylan #34: PiZero Bluetooth not starting

ChasinSpin, 2020-02-22:
There appears to be a race condition On Pi Zero W and Buster.

If:
There is a wifi network configured, but the device is moved to somewhere that the network is not available OR no network is configured (i.e. all the networks are deleted with nmcli), then
the nymea-networkmanager doesn't recognize there's no network on startup.   (everything with default configuration)

In this state, restarting the service after the service starts up during boot fixes it.

However it doesn't happen if the network is available but has the wrong password, in this situation it doesn't work.

To solve this, I needed to delay the nymea-networkmanager.service startup with:
ExecStartPre=/bin/sleep 60

Great product BTW

mzanetti, 2020-02-23:
Thanks for the report. I believe this is the race condition we found and fixed recently:

https://github.com/nymea/libnymea-networkmanager/pull/5

The fix is not released to stable yet. You can test if this fixes your issue by upgrading to the experimental silo. For that, add this line

```deb http://ci-repo.nymea.io/experimental-silo buster rpi```

to `/etc/apt/sources.list.d/nymea.list` and apt-get update and upgrade.

mzanetti, 2020-03-05:
Is this still an issue with the latest release?

closed 2020-12-13 by mzanetti