nymea/berrylan #35: Issue with piZero install

kkobylinski, 2020-02-24:
Hi,
I'm getting the issue below:

pi@raspberrypi:~ $ sudo apt-key adv --keyserver keyserver.ubunt.com:80 --recv-keys A1A19ED6
Executing: /tmp/apt-key-gpghome.ylrW6iNfBM/gpg.1.sh --keyserver keyserver.ubunt.com:80 --recv-keys A1A19ED6
gpg: keyserver receive failed: **No keyserver available**

Thanks for help in advance!

mzanetti, 2020-02-24:
Your --keyserver argument seems wrong...

a) it's `ubuntu`, not `ubunt`
b) I don't think it supports passing the port :80 there...

it works for me with
```
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys A1A19ED6
```

mzanetti, 2020-02-24:
Did you copy this like that from one of our documentation pages? If so, please send me the link so it can be fixed

krissyKris, 2020-02-24:
I think it is time to sleep for me :) thanks for spotting such a silly mistake, documentation is fine!

closed 2020-02-24 by mzanetti