nymea/berrylan #38: Cannot open Access Point with BerryLan App

blasco991, 2020-04-20:
Hì, i have a Raspberry Pi 3 Model A Plus Rev 1.0 

With just the clean image downloaded from http://www.berrylan.org/ i can use the BerryLan mobile App and successfully connect to an existing WiFi, but when i try to create an Access Point with the relative button on the App, i give it the SSID and PSW but look at the syslog on the raspberry i get this error log:

pi@raspberrypi:~ $ sudo nymea-networkmanager -m start
 I | Application: Using configuration file from: "/etc/nymea/nymea-networkmanager.conf"
 I | Application: =====================================
 I | Application: Starting nymea-networkmanager "0.5.2+202003241320~buster+rpi1"
 I | Application: =====================================
 I | Application: Advertising name: "BT WLAN setup"
 I | Application: Platform name: "nymea-box"
 I | Application: Mode: Core::ModeStart
 I | Application: Timeout: 60
 W | NymeaService: Invalid D-Bus HardwareManager interface.
 W | NymeaService: Could not init nymea D-Bus services
 I | Application: Networkmanager is now available.
 I | Application: Starting the Bluetooth service because of "start" mode.
 W | NymeaService: Could not enable/disable bluetooth hardware resource. D-Bus interface not available.
 W | qt.bluetooth: Using BlueZ kernel ATT interface
 I | Application: Bluetooth server started
 I | Application: Bluetooth client connected
 W | qt.bluetooth.bluez: sending error response; request: 16 handle: 51 code: 10
 W | qt.bluetooth.bluez: sending error response; request: 8 handle: 1 code: 10
 W | qt.bluetooth.bluez: sending error response; request: 8 handle: 26 code: 10
 W | qt.bluetooth.bluez: sending error response; request: 8 handle: 39 code: 10
 W | NetworkManager: "org.freedesktop.NetworkManager.Settings.Connection.InvalidProperty" "802-11-wireless-security.proto: can't set property of type 'GStrv' from value of type 's'"

t-mon, 2020-04-20:
Hi @blasco991 

thanks for reporting this, I noticed this behavior already and I'm working on a fix.
Looks like on the buster version of the network-manager something changed on the dbus side and should be quiet easy to fix.

I'll let you know once I have a build to test if it's working again.

mzanetti, 2020-04-24:
The bug has been fixed in libnymea-networkmanager and is in the experimental repository now.

Thanks for reporting.

blasco991, 2020-05-06:
> The bug has been fixed in libnymea-networkmanager and is in the experimental repository now.
> 
> Thanks for reporting.

thank you, when should this fix be available on the packet that you install with apt-get?

mzanetti, 2020-05-06:
It will be released with nymea 0.21. Should be in a few weeks, two maybe. We just released 0.20 when you reported this... Needs to go through the testing repository and QA tested still.

If you're feeling adventurous, you can also enable the experimental repository and test it right away by adding this to /etc/apt/sources.list.d/nymea.list
```
deb http://ci-repo.nymea.io/experimental-silo buster rpi
```
then you'll get it with apt-get upgrade

closed 2020-06-09 by nymea-jenkins