nymea/berrylan #40: Embedding berrylan mobile app into PWA

ivelin, 2020-06-26:
Awesome project. Congrats!

Has there been a discussion about supporting PWA as an additional and alternative client to native mobile apps?  
https://www.hackster.io/patricia2/offline-pwa-with-bluetooth-le-capabilities-77cbba

PWAs are becoming popular as a faster way to deliver quality UI to multiple devices. Ambianic UI is a PWA and WiFi onboarding for the RPI edge device is an open topic we are investigating. If there is interest in supporting PWA, we can potentially collaborate on the implementation. See ambianic/ambianic-ui/issues/397

mzanetti, 2020-06-27:
Hi. I for one haven't looked into this and probably won't be doing so anytime soon. But if someone wants to get going with this, here's the documentation of the Bluetooth GATT protocol that BerryLan uses: 

https://github.com/nymea/nymea-networkmanager/

I'll be happy to support on any upcoming questions on the protocol.

ivelin, 2020-06-29:
Thank you for the response @mzanetti . We will take a look at the protocol and see how we can contribute a PWA implementation.

keriat, 2020-08-02:
Hey, @ivelin!

Did you succeed with the PWA? Or you've assured correct way to connect with Bluetooth GATT protocol in js?

ivelin, 2020-08-04:
Hi @keriat . The project is in progress. You can see where we are at and join in if you like:
https://github.com/ambianic/ambianic-ui/issues/397

@BKristenssonAlfsson is leading the charge on the UI/JS side and @muca is covering the RPI side.

keriat, 2020-10-06:
@ivelin hey-hey! 👋 
I've successfully make the connection with the Raspberry Pi work.

 And the whole setup of it

ivelin, 2020-10-06:
Congrats, @keriat ! Is your work under a public repo?

spacewalkingninja, 2020-10-11:
@keriat Hey! I am also wondering if it is public repo. Can you please post it in a public repo. Thx! Big Hug & Kiss!

keriat, 2020-10-27:
@spacewalkingninja hey! Yeah, going to publish it soon in form of an NPM package ;)

So stay tuned 😉

spacewalkingninja, 2020-10-27:
Oh friend, I am very tuned! Cheers!

osarhan, 2023-03-01:
@keriat this is a few years old now. Curious if you ever got your working example into a public repo :)