nymea/berrylan #42: update/upgrade of latest BerryLan breaks Bluetooth

davidallenmann, 2020-08-13:
Started with latest build of BerryLan (2020-07-22).

`sudo nano /etc/nymea/nymea-networkmanager.conf`
```
Mode=start
Timeout=1200
AdvertiseName=MyDevice
PlatformName=MyDevice
ButtonGpio=15
```
reboot

Can see MyDevice on BerryLan app on iOS iPhone.

Update Pi with:
```
sudo apt-get update
sudo apt-get upgrade
```

MyDevice is no longer visible on BerryLan app.

This worked about two weeks ago. So, something in Aug 7 build may be causing issue.

davidallenmann, 2020-08-13:
bluez after upgrade shows
`sudo apt list --installed | grep bluez

bluez-firmware/testing,now 1.2-4+rpt3 all [installed,automatic]
bluez/testing,now 5.50-1.2~deb10u1+rpt2 armhf [installed,automatic]
`

Making changes to privacy seems to have fixed advertising issue.

https://unix.stackexchange.com/questions/382031/bluez-error-setting-privacy-on-raspbian

davidallenmann, 2020-08-13:
Holding bluez package prior to upgrade prevents issue.

`sudo apt-mark hold bluez`

After upgrade
```
bluez-firmware/testing,now 1.2-4+rpt3 all [installed,automatic]
bluez/now 5.50-1.2~deb10u1+rpt1 armhf [installed,upgradable to: 5.50-1.2~deb10u1+rpt2]
```

mzanetti, 2020-08-13:
Thank you for reporting this.

mzanetti, 2020-08-26:
I've just tested the latest release candidate for the berrylan image from https://downloads.nymea.io/images/berrylan/landing/

In there everything seems to work fine with those versions:

```
pi@raspberrypi:~ $ apt-cache policy bluez-firmware
bluez-firmware:
  Installed: 1.2-4+rpt4
  Candidate: 1.2-4+rpt4
  Version table:
 *** 1.2-4+rpt4 500
        500 http://archive.raspberrypi.org/debian buster/main armhf Packages
        100 /var/lib/dpkg/status
     1.2-4 500
        500 http://raspbian.raspberrypi.org/raspbian buster/non-free armhf Packages
pi@raspberrypi:~ $ apt-cache policy bluez 
bluez:
  Installed: 5.50-1.2~deb10u1+rpt2
  Candidate: 5.50-1.2~deb10u1+rpt2
  Version table:
 *** 5.50-1.2~deb10u1+rpt2 500
        500 http://archive.raspberrypi.org/debian buster/main armhf Packages
        100 /var/lib/dpkg/status
     5.50-1.2~deb10u1 500
        500 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages
```
So to me this seems like an issue in upstream bluez/bluez-firmware which might happened in the bluez-firmware-1.2-4+rpt3. The version with +rpt4 seems to be fine again.

Closing this. Please feel free to reopen if you still experience problems.

closed 2020-08-26 by mzanetti