nymea/berrylan #44: Question

umsaenga, 2020-10-09:
Hi, 
Can your app be modified to also send commands to the rpi as a controller? Say i have some python and shell commands, i want to use app to execute them. thank you

mzanetti, 2020-10-11:
Hi.

Perhaps you might find nymea:app useful. Download the full nymea featured RPi image and nymea:app. With that you can use the BerryLan features and also set up arbitrary commands to be executed on the RPi (besides much more).

umsaenga, 2020-10-18:
berryLan works fine when I install on existing raspbian lite. It doesn't when I use your image. 

nymea:app with your nymea raspbian image doesn't work. It doesn't find the RPi on bluetooth to set up the wifi.

mzanetti, 2020-10-18:
Are you using a Raspberry Pi 4?

umsaenga, 2020-10-19:
yes

mzanetti, 2020-10-19:
I've found an issue which affects only the rpi4.

This should fix it for you I believe: https://github.com/nymea/nymea-networkmanager/issues/36

mzanetti, 2020-10-21:
please try the latest experimental images which should fix this:

BerryLan image: https://downloads.nymea.io/images/berrylan/experimental/
nymea image: https://downloads.nymea.io/images/raspberrypi/experimental/

umsaenga, 2020-10-24:
okay, thank you, will test and revert back with results.

umsaenga, 2020-11-05:
I test your berrylan image and it seems to work `berrylan-image-raspbian-buster-2020-10-27.zip | 2020-10-27 22:32 | 496M |  `

I test manual installation as per the current readme and it gets stuck on `Setting up libqt5core5a:armhf (5.11.3+dfsg1-1+rpi1+deb10u4) ...`

mzanetti, 2020-11-05:
Hmm... Not sure why the installation got stuck. Perhaps it overheatet? The Original Raspberry Pi 4 case is known to cause such troubles.

In any case, we've just released the fix to stable right now. So if you tried the manual installation 3h ago you wouldn't still get the fixed version. Now you should get that one.

Also the stable berrylan image should be fixed now to work reliably on the RPi4. Thanks for this report.

I'll close this issue now. If you still experience any troubles, please open a new one with an appropriate title/description as this report already mixes 2 different topics. Thanks a lot.

closed 2020-11-05 by mzanetti