nymea/berrylan #45: Question: Berrylan Android app does not list Wifi when Channel set to "Auto" in router??

raomaneesh, 2020-10-18:
Hi,
I have recently replaced my router with new dual channel router (5G/2.4G). This router has **Channel** configuration set to **Auto**  and changing channel is disabled. Due this the BerryLan App does not list my wifi and I am not able to connect my Rpi to wifi through it. Earlier with my old router I use to set **Channel = 11** to make it work. So my question is how to make Rpi work for **Channel  = Auto**  using berrylan app?

raomaneesh, 2020-10-23:
Discussing the same on forum:
https://forum.nymea.io/t/berrylan-app-does-not-list-wifi-when-channel-set-to-auto-in-router/381

closed 2020-10-23 by raomaneesh

mzanetti, 2020-10-23:
Thanks. And sorry for not replying here earlier...