nymea/berrylan #47: Installed but not showing

rijotech, 2020-12-12:
I installed on DietPi_RPi-ARMv8-Buster fallowing the installation process but using the buster repo 
deb http://repository.nymea.io buster main
I see the service nymea-networkmanager running in the background, but it doesn't show on the berrylan app.
There is no dhcpd installed, but still nothing shows in the app.

mzanetti, 2020-12-12:
Please check the logs of nymea-networkmanager

    sudo journalctl -u nymea-networkmanager

rijotech, 2020-12-12:
-- Logs begin at Thu 2019-02-14 05:11:58 EST, end at Sat 2020-12-12 18:40:08 EST. --
Dec 12 18:18:54 raspled systemd[1]: Started Daemon for nymea to configure wifi network using a Bluetooth LE connection.
Dec 12 18:18:55 raspled nymea-networkmanager[493]:  I | Application: Using configuration file from: "/etc/nymea/nymea-networkmanager.conf"
Dec 12 18:18:55 raspled nymea-networkmanager[493]:  I | Application: =====================================
Dec 12 18:18:55 raspled nymea-networkmanager[493]:  I | Application: Starting nymea-networkmanager "0.5.5+202011171557~buster1"
Dec 12 18:18:55 raspled nymea-networkmanager[493]:  I | Application: =====================================
Dec 12 18:18:55 raspled nymea-networkmanager[493]:  I | Application: Advertising name: "BT WLAN setup"
Dec 12 18:18:55 raspled nymea-networkmanager[493]:  I | Application: Platform name: "nymea-box"
Dec 12 18:18:55 raspled nymea-networkmanager[493]:  I | Application: Mode: Core::Mode(ModeOffline)
Dec 12 18:18:55 raspled nymea-networkmanager[493]:  I | Application: Timeout: 60
Dec 12 18:18:55 raspled nymea-networkmanager[493]:  W | NymeaService: Invalid D-Bus HardwareManager interface.
Dec 12 18:18:55 raspled nymea-networkmanager[493]:  W | NymeaService: Could not init nymea D-Bus services
Dec 12 18:18:55 raspled nymea-networkmanager[493]:  I | NetworkManager: Starting the network manager.
Dec 12 18:18:55 raspled nymea-networkmanager[493]:  I | NetworkManager: Initializing network manager
Dec 12 18:18:55 raspled nymea-networkmanager[493]:  I | NetworkManager: DBus interface created successfully "/org/freedesktop/NetworkManager"
Dec 12 18:18:55 raspled nymea-networkmanager[493]:  I | NetworkManager: Reading initial properties...
Dec 12 18:18:55 raspled nymea-networkmanager[493]:  I | NetworkManager: Version: "1.14.6"
Dec 12 18:18:55 raspled nymea-networkmanager[493]:  I | NetworkManager: State changed: "NetworkManagerStateConnectedGlobal"
Dec 12 18:18:55 raspled nymea-networkmanager[493]:  I | Application: Not advertising bluetooth because we are online and we are running in Cor
e::Mode(ModeOffline)
Dec 12 18:18:55 raspled nymea-networkmanager[493]:  I | NetworkManager: Networking enabled
Dec 12 18:18:55 raspled nymea-networkmanager[493]:  I | NetworkManager: Wireless networking enabled
Dec 12 18:18:55 raspled nymea-networkmanager[493]:  I | NetworkManager: Get available devices
Dec 12 18:18:55 raspled nymea-networkmanager[493]:  I | NetworkManager: [+] NetworkDevice("lo" - "Generic", "NetworkDeviceStateUnmanaged")
Dec 12 18:18:55 raspled nymea-networkmanager[493]:  I | NetworkManager: [+] WiredNetworkDevice("eth0", "DC:A6:32:BF:D7:19", 0 [Mb/s], true, "N
etworkDeviceStateUnmanaged")
Dec 12 18:18:55 raspled nymea-networkmanager[493]:  I | NetworkManager: Load connection list
Dec 12 18:18:55 raspled nymea-networkmanager[493]:  I | NetworkManager: The network manager is now available
Dec 12 18:18:55 raspled nymea-networkmanager[493]:  I | Application: Networkmanager is now available.
Dec 12 18:18:55 raspled nymea-networkmanager[493]:  I | Application: Not advertising bluetooth because we are online and we are running in Cor
e::Mode(ModeOffline)
Dec 12 18:18:55 raspled nymea-networkmanager[493]:  I | NetworkManager: Network manager initialized successfully.
Dec 12 18:18:55 raspled nymea-networkmanager[493]:  I | NetworkManager: Starting initial wireless network scan...

mzanetti, 2020-12-12:
As the log says:

    Application: Not advertising bluetooth because we are online and we are running in Core::Mode(ModeOffline)

It will only advertise itself when the device is offline unless it is set to "alyways" in the configuration

rijotech, 2020-12-12:
I think I know what the problem is

rijotech, 2020-12-12:
> As the log says:
> 
> ```
> Application: Not advertising bluetooth because we are online and we are running in Core::Mode(ModeOffline)
> ```
> 
> It will only advertise itself when the device is offline unless it is set to "alyways" in the configuration

Yep Thanks

rijotech, 2020-12-12:
I'm sorry

mzanetti, 2020-12-13:
I'd say closing this is fine...

closed 2020-12-13 by mzanetti

rijotech, 2020-12-13:
Sorry I closed it before testing, I disconnected the Lan cable but still doesn't work.
I changed the config to always and still the same.
This is what I'm getting now.

-- Logs begin at Thu 2019-02-14 05:11:58 EST, end at Sat 2020-12-12 18:55:31 EST. --
Dec 12 18:46:30 raspled systemd[1]: Started Daemon for nymea to configure wifi network using a Bluetooth LE connection.
Dec 12 18:46:31 raspled nymea-networkmanager[692]:  I | Application: Using configuration file from: "/etc/nymea/nymea-networkmanager.conf"
Dec 12 18:46:31 raspled nymea-networkmanager[692]:  I | Application: =====================================
Dec 12 18:46:31 raspled nymea-networkmanager[692]:  I | Application: Starting nymea-networkmanager "0.5.5+202011171557~buster1"
Dec 12 18:46:31 raspled nymea-networkmanager[692]:  I | Application: =====================================
Dec 12 18:46:31 raspled nymea-networkmanager[692]:  I | Application: Advertising name: "BT WLAN setup"
Dec 12 18:46:31 raspled nymea-networkmanager[692]:  I | Application: Platform name: "nymea-box"
Dec 12 18:46:31 raspled nymea-networkmanager[692]:  I | Application: Mode: Core::Mode(ModeAlways)
Dec 12 18:46:31 raspled nymea-networkmanager[692]:  I | Application: Timeout: 60
Dec 12 18:46:31 raspled nymea-networkmanager[692]:  W | NymeaService: Invalid D-Bus HardwareManager interface.
Dec 12 18:46:31 raspled nymea-networkmanager[692]:  W | NymeaService: Could not init nymea D-Bus services
Dec 12 18:46:31 raspled nymea-networkmanager[692]:  I | NetworkManager: Starting the network manager.
Dec 12 18:46:31 raspled nymea-networkmanager[692]:  I | NetworkManager: Initializing network manager
Dec 12 18:46:31 raspled nymea-networkmanager[692]:  I | NetworkManager: DBus interface created successfully "/org/freedesktop/NetworkManager"
Dec 12 18:46:31 raspled nymea-networkmanager[692]:  I | NetworkManager: Reading initial properties...
Dec 12 18:46:31 raspled nymea-networkmanager[692]:  I | NetworkManager: Version: "1.14.6"
Dec 12 18:46:31 raspled nymea-networkmanager[692]:  I | NetworkManager: State changed: "NetworkManagerStateConnectedGlobal"
Dec 12 18:46:31 raspled nymea-networkmanager[692]:  I | NetworkManager: Networking enabled
Dec 12 18:46:31 raspled nymea-networkmanager[692]:  I | NetworkManager: Wireless networking enabled
Dec 12 18:46:31 raspled nymea-networkmanager[692]:  I | NetworkManager: Get available devices
Dec 12 18:46:31 raspled nymea-networkmanager[692]:  I | NetworkManager: [+] NetworkDevice("lo" - "Generic", "NetworkDeviceStateUnmanaged")
Dec 12 18:46:31 raspled nymea-networkmanager[692]:  I | NetworkManager: [+] WiredNetworkDevice("eth0", "DC:A6:32:BF:D7:19", 0 [Mb/s], false, "NetworkDeviceStateUnmanaged")
Dec 12 18:46:31 raspled nymea-networkmanager[692]:  I | NetworkManager: Load connection list
Dec 12 18:46:31 raspled nymea-networkmanager[692]:  I | NetworkManager: The network manager is now available
Dec 12 18:46:31 raspled nymea-networkmanager[692]:  I | Application: Networkmanager is now available.
Dec 12 18:46:31 raspled nymea-networkmanager[692]:  I | Application: Starting the Bluetooth service because of "always" mode.
Dec 12 18:46:31 raspled nymea-networkmanager[692]:  W | Application: Could not start services. There is no wireless device available.
Dec 12 18:46:31 raspled nymea-networkmanager[692]:  I | NetworkManager: Network manager initialized successfully.
Dec 12 18:46:31 raspled nymea-networkmanager[692]:  I | NetworkManager: Starting initial wireless network scan...

mzanetti, 2020-12-13:
Oh well... now the log says

    Application: Could not start services. There is no wireless device available.

So seems like there is no WiFi available on your hardware...

Please note that this is not a support platform but a bug tracker for actual bugs in the software. You may want to use forum.nymea.io to ask for help on setting up things.

rijotech, 2020-12-13:
> Oh well... now the log says
>
> ```
> Application: Could not start services. There is no wireless device available.
> ```
>
> So seems like there is no WiFi available on your hardware...
>
> Please note that [...]

Ok thanks for your response.