nymea/berrylan #50: Problem

WerVa, 2021-01-31:
When i start nymea-networkmenager no boot 

XXX@XXX:~$ systemctl status nymea-networkmanager
● nymea-networkmanager.service - Daemon for nymea to configure wifi network using a Bluetooth LE connection
   Loaded: loaded (/lib/systemd/system/nymea-networkmanager.service; enabled; vendor preset: enabled)
   Active: inactive (dead)
     Docs: https://github.com/nymea/nymea-networkmanager

WerVa, 2021-02-01:
i found this in log  
W | NetworkManager: Scan error: "org.freedesktop.NetworkManager.PermissionDenied" "org.freedesktop.NetworkManager.network-control request failed: not authorized"

What is should to do

t-mon, 2021-02-01:
Hello @WerVa 
according to the systemctl status, the daemon seems to be enabled and should autostart during boot.

Maybe the logs tell us what's going wrong.
- restart the system
- check with `journalctl -b -u nymea-networkmanager` the logs since boot, maybe there is something going wrong.
- Is the daemon up to date? you can check the version with `apt policy nymea-networkmanager`, since we had a start up problem few months ago.

If the daemon is not starting at all, there are no logs and something is wrong with the systemd service.
If there are logs, they hopefully give us a hint why the daemon is not able to start during boot.

WerVa, 2021-02-01:
journalctl -b -u nymea-networkmanager
-- Logs begin at Thu 2019-02-14 10:11:59 GMT, end at Mon 2021-02-01 09:12:15 GMT. --
-- No entries --

> apt policy nymea-networkmanager
nymea-networkmanager:
  Installed: 0.5.5+202011171741~buster+rpi1
  Candidate: 0.5.5+202011171741~buster+rpi1
  Version table:
 *** 0.5.5+202011171741~buster+rpi1 500
        500 http://ci-repo.nymea.io/experimental-silo buster/rpi armhf Packages
        100 /var/lib/dpkg/status

WerVa, 2021-02-01:
I can run manualy only: 

BlackMirror@MyBlackMirror:~$ systemctl start  nymea-networkmanager
==== AUTHENTICATING FOR org.freedesktop.systemd1.manage-units ===
Authentication is required to start 'nymea-networkmanager.service'.
Authenticating as: ,,, (BlackMirror)
Password: 
==== AUTHENTICATION COMPLETE ===
BlackMirror@MyBlackMirror:~$ systemctl status  nymea-networkmanager
● nymea-networkmanager.service - Daemon for nymea to configure wifi network using a Bluetooth LE connection
   Loaded: loaded (/lib/systemd/system/nymea-networkmanager.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2021-02-01 09:45:45 GMT; 5s ago
 Main PID: 1394 (nymea-networkma)
    Tasks: 2 (limit: 1330)
   CGroup: /system.slice/nymea-networkmanager.service
           └─1394 /usr/bin/nymea-networkmanager -d

WerVa, 2021-02-01:
i add this to supervisor 

[program:nymea-networkmenager]
command=/usr/bin/nymea-networkmanager -d
autostart=true
autorestart=true

and working... you can close problem

t-mon, 2021-02-01:
So the problem was concurring process manager?
You use supervisor and systemd?
Can you describe the environment, just in case somebody else runs into this issue?

WerVa, 2021-02-01:
this is the environment for my project, it’s extremely limited, so I dont think anybody should have this problem

closed 2021-02-01 by WerVa