nymea/berrylan #55: connecting with a fixed ip address without dhcp available

rdesanctis, 2021-05-03:
ISSUE: always asking for a password if router is not running DHCP server (fixed ip address)

CASE DETAIL:
i use multiple wifi routers with fixed ip address without dhcp service, each time berrylan try to configure lan it ask for password, and after some time says wrong password (probably due to non-receiving ip address because routers are configured without DHCP service enabled). Each router had their own a different network , i can not fix a ip address by default on the raspi side because dhcpd.conf only support 1 router at time

SUGESTED SOLUTION: if berrylan connect to Wifi and does not receive a ip address, then ask for a ip address instead of asking again for a password

mzanetti, 2021-05-03:
This requires support in https://github.com/nymea/nymea-networkmanager

t-mon, 2021-05-03:
We should do those changes using API V2, which is under development.