nymea/berrylan #67: E: Unable to locate package nymea-networkmanager

dakschnitzer, 2021-12-13:
Hi - I am working off of a clean install of buster. I've followed the instructions to install the BerryLan networkmanager manually on an existing image. Unfortunately, I get an error, "Unable to locate package nymea-networkmanager." 

I've tried changing "rpi" to "main" in the echo command line, but that doesn't seem to make a difference.

What am I doing wrong here?

mzanetti, 2021-12-13:
Component "rpi" should be ok, "main" should work too if you're on a RPi 2, 3 or 4. For a Pi 0 or 1 only "rpi" would work.

Not sure what you're doing wrong. Perhaps forgot `sudo apt-get update` after editing the source.list file?

dakschnitzer, 2021-12-13:
Thanks so much. I had just run `sudo apt-get update` after installing the OS and didn't realize I needed to run it again after editing the source.list file for the berrylan install.

closed 2021-12-13 by dakschnitzer