nymea/berrylan #73: Bluetooth advertisement

rnehrboss, 2022-06-30:
Does anyone know if the advertisement of bluetooth only lasts for a short time after boot?   It seems like they dissapear from the android app, even though they aren't connected to wifi.

Also we changed the advertisment name, and it seems like it first says "raspberry pi" for a short time, then goes to BT-WLAN, before finally going to the name we changed it to.   Is this normal?

Thanks!

t-mon, 2022-07-01:
The advertisment behavior depends on the mode you configured for the `nymea-networkmanager`. 
There is also a boot mode which enabled the advertisement for a certain time after boot and disables afterwards.
Default is the offlin mode, which should restart th bluetooth server if you are offline (independet if lan or wlan). For testing you could try to use the always mode, which should advertise always...

Regarding the advertisement name: we noticed that behavior too, and it comes from the mix of classic bluetooth advertisement and low energy. If you see the "raspberry pi" name first, that is you pretty name you can change in the `/etc/machine-info` -> `PRETTY_HOSTNAME=bluetooth-name`.

If your host is discoverable, your phone shows the classic bluetooth name (pretty-hostname), once the nymea-networkmanager starts the service, a bluetooth LE server will be started and begins to advertise as LE device. This leads to the behaviour you noticed, first the name of the service is the hostname, then th package containing the server name arrives and the phone updates the name...

TechSgin, 2022-09-19:
We are experiencing issues with berryLan on android (we didn't try iOS) .
It seems like the "phone" (android Samsung A33, LG V20 and few others) list the Bluetooth "entry" but it is not listed on the berryLan app.

**we have tried:**
1. filtering enabled \ disabled
2. change the AdvertiseName in /etc/nymea/nymea-networkmanager.conf (only 8 chars)
3. change the timeout in /etc/nymea/nymea-networkmanager.conf
4. change the mode offline, start and always
5. change the host name to match the AdvertiseName  (only 8 chars)
6. we have tried to re-install berryLan on the phone 
7. we have updated \ upgraded the service on the pi 

**all with no success**

When looking at the logs "journalctl -u nymea-networkmanager.service" all seems to be ok .. 
We will be more than happy to get some help here .. 

Thanks In Advance Oren

mzanetti, 2022-09-20:
Are location services enabled on the Phone? For some reason Android requires Location Services to be enabled in order to deliver Bluetooth search results.

TechSgin, 2022-09-20:
> 

Thank You Very Much @mzanetti 
Amazing, enabling "Location Services" solved the issue and we can now see the device on our phones.
* we have spend few hours for the last couple of days, we couldn't figure this out.

closed 2022-09-20 by mzanetti