nymea/berrylan #79: Unable to install on PI Zero 2W

Palmeta, 2022-11-19:


closed 2022-11-20 by mzanetti

Palmeta, 2022-11-23:
Has this problem been fixed?

Palmeta, 2022-11-23:
Selecting previously unselected package qttranslations5-l10n.
Preparing to unpack .../11-qttranslations5-l10n_5.15.2-2_all.deb ...
Unpacking qttranslations5-l10n (5.15.2-2) ...
Setting up libdouble-conversion3:armhf (3.1.5-6.1) ...
Setting up libpcre2-16-0:armhf (10.36-2+deb11u1) ...
Setting up qttranslations5-l10n (5.15.2-2) ...
Setting up shared-mime-info (2.0-1) ...
Setting up libqt5core5a:armhf (5.15.2+dfsg-9+rpi1) ...
Setting up libqt5dbus5:armhf (5.15.2+dfsg-9+rpi1) ...
Setting up libqt5network5:armhf (5.15.2+dfsg-9+rpi1) ...
Setting up libnymea-gpio:armhf (1.5.1+202211071351~bullseye+rpi1) ...
Setting up libqt5bluetooth5-bin (5.15.2-2) ...
Setting up libqt5bluetooth5:armhf (5.15.2-2) ...
Setting up libnymea-networkmanager (1.5.1+202211071351~bullseye+rpi1) ...
Setting up nymea-networkmanager (1.5.1+202211071624~bullseye+rpi1) ...
Created symlink /etc/systemd/system/graphical.target.wants/nymea-networkmanager.service → /lib/systemd/system/nymea-networkmanager.service.
Processing triggers for man-db (2.9.4-2) ...

Palmeta, 2022-11-23:
It always crashes here

mzanetti, 2022-11-23:
The given log does not show any problem. If it just stalls here, perhaps the SD card is corrupted or the power supply is too weak, or the system is running out of memory. Can't say more with the given information. In any case, unlikely it is actually related to nymea-networkworkmanager itself.

Palmeta, 2022-11-23:
> Hi, thanks. I've tried multiple cards. Everything runs fine until I try to
> install nymea. I've tried with your pre-setup image but it just fails to
> boot. If I try setting it up through your repository it crashes at the
> point I showed you.
> And I mean really crashes no more ssh console to it. Then I unplug it. Re
> plug and it boots ssh ok. The pi doesn't show up in your android app. When
> I disable dhcpd in an attempt to see if it works, dead again! This time
> possibly since it is headless and to get to it I have to re-flash the card.

Do you have a PI zero 2W you can try it on?

Palmeta, 2022-11-23:
I'm fully updated, running bullseye.
Let me say that your work is extremely valuable and the only such piece of
software around. Bluetooth on the pi is mayhem ever since javax.bluetooth
was deprecated and abandoned without any replacement.

Palmeta, 2022-11-23:
Sorry read berrylan where I wrote nymea

mzanetti, 2022-11-23:
I don't have a pi zero 2 W... Only a first gen.

Is it possible that it just drops the WiFi connection while it's installing network-manager? Did you try using a serial port console for the installation?
Perhaps wait for a bit and check if the Bluetooth service comes up to configure wifi again. At least the first gen Pi Zero is really slow, make sure to give it enough time...

Palmeta, 2022-11-24:
This new pi zero is an impressive little machine. It is a four core and its
only weakness is its 512 mb of ram but I don't run it with a gui so there
is no problem there. Apart from that it is not shy of a pi3.
The only way to connect to it is via wifi or bluetooth as it does not have
lan. it does have a micro hdmi port for a monitor but i haven't tried that
yet so its only ssh. Berrylan would be a killer for it because if you would
like to take it somewhere and connect it without hauling your own router or
flashing the card on site with local credentials to wifi it would be the
way to go. I've been trying to create my own solution but it's a hefty job
since I am a Java guy and java support for bluetooth is dead.

mzanetti, 2022-11-24:
We're using nymea-networkmanager on a multitude of devices. It works fine. On the release of the latest stable image I have tested it on a RPi3. We're building images that include installing nymea-networkmanager on a daily basis. So... not sure what to tell... to me it looks like that your wifi connection drops when you install nymea-networkmanger, which pulls in network-manager which will take over the wifi configuration. so that's expected. Either preinstall it in your image before flashing, or use a serial port console (or display and keyboard) for the installations instead of ssh.

simonseres, 2022-11-24:
@Palmeta just tested a Pi Zero 2 W with actual BerryLan image _2022-11-08-berrylan-bullseye-lite.img_ and WiFi Setup works as expected.
[Download image](https://downloads.nymea.io/images/berrylan/latest)

Palmeta, 2022-11-24:
Great news! I burned that image and it worked like a charm!
Thanks!

Palmeta, 2022-11-24:
But the time is messed up. NTP is not there.