nymea/berrylan #82: After reboot, wifi interface doesn't come up, bluetooth doesn't find device

DFC72, 2023-07-16:
I've installed onto the following platform:
Pi Zero W
Raspbian 32-bit Lite

Performed a fresh install of the OS.
_sudo apt update && sudo apt upgrade -y_

added repository like this:
_echo "deb http://repository.nymea.io bullseye rpi" | sudo tee /etc/apt/sources.list.d/nymea.list
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-key A1A19ED6_

updated apt:
_sudo apt update_

installed nymea-networkmanager:
_sudo apt install nymea-networkmanager dirnmgr_

No issues there...

Disabled dhcpd:
_sudo systemctl disable dhcpcd_

Rebooted:
_sudo reboot_

After reboot, the wlan0 interface doesn't come up on the original SSID, and when I try to connect to the Pi using the iPhone app, it doesn't find the pi. I've also tried disabling the filter, but there are a ton of BT devices with no name, and no way to see which may be the Pi.

Where to go from here?

How do I re-enable the wlan0 interface on the Pi?

mzanetti, 2023-07-17:
It should work just fine, but note that the Pi Zero W is really slow, it may take several minutes until its fully bootet and visible on Bluetooth.

DFC72, 2023-07-17:
I'll try again and wait longer before I reboot! I've got a Zero 2 W coming for my prototype version I'm sending out, so hopefully that will work.

Alternatively, I could build on a RPi4, and it should be able to proof the build and then just bring it up on the Zero after switching in the SD card?

mzanetti, 2023-07-17:
Yeah, should work I suppose.

The Zero 2 W is definitely much faster than the v1, yes.

DFC72, 2023-07-17:
Ok, this is weird now - I pulled the sd and pout it in the RPi 4B and it works fine. I delete the SSId profile by running sudo nmcli con del 'CONNECTION_NAME' and the wifi connection drops. I open the app on my iPhone and it finds it right away. I connect WiFi using the app and everything works fine.

I try the same thing on the Pi Zero W - waited looking at the app on my phone and it didn't show up - even after 15m....

Is there some difference in hardware between the two that would prevent this from working on the Zero W? (BT4 on the ZeroW  vs BT5 on the Pi4B?)

DFC72, 2023-07-17:
I wanted to use the smaller dimensions of the Pi Zero (and 2) for a product I'm prototyping - The 4B is overkill for me... plus expensive and hard to find currently. If I could find a supply if 1/2GB models that may be an option, but still 3-4 times more expensive than the Zero platform. Would the Pi Zero 2W be supported in the future with the 4-core cpu?

I'll try the swap file increase to 1GB as it's still at the default of 100MB- I have direct access to the device as it's plugged into a monitor and keyboard currently while I'm prototyping... Would that give me access to the boot logs?

As for the nymea logs, I looked for them with fzf but couldn't find them - could you pls point me in the right direction? Do I need to use journalctl:
_sudo journalctl -u nymea-networkmanager_ ?

I also found this thread with someone else having the same issue a few years ago - https://github.com/nymea/nymea-networkmanager/issues/29

Last update was as you see - Could be a race condition with some of the later builds and related to arm6 vs arm7?

My Zero 2 W should be arriving this week so I'll see if the newer cpu fixes this issue and report back unless you have a further idea based on the older post ?

mzanetti, 2023-07-17:
Yes, the Zero 2 W (which is comparable to a RPi 3 in terms of horse power) should be good for another while still.

DFC72, 2023-07-19:
Just received my Pi Zero 2W - Can confirm that this issue does not occur on the newer Zero.