nymea/berrylan #84: Login failed - Access denied

vfricke, 2023-08-09:
After sucessful setup using Android app BerryLan a ssh login failed to the device. User pi password raspberry resulted in message "Access denied". BerryLan image is being used as base image. Any idea of the problem? Thanks

nymea-jenkins, 2023-08-14:
Hi!
Just out of my mind: BerryLan is a slightly modificated Rasperry OS Image. I think SSH has to be explicity enabled. User and password should be the same as any other Raspberry OS.

vfricke, 2023-09-11:
> Hi! Just out of my mind: BerryLan is a slightly modificated Rasperry OS Image. I think SSH has to be explicity enabled. User and password should be the same as any other Raspberry OS.
> […](#)

Correct. However BerryLan is designed for head-less operation. When you use Rasperry OS with SSH disabled it does not make much such. You want to have an image which provides you a head-less login with SSH enabled, right?

mzanetti, 2023-09-11:
The plain BerryLan image is intentionally designed with minimal modifications to the upstream Raspberry pi OS image, that is, just adding the nymea-networkmanager and nothing else. User can then set up their services and then move the RPi between wireless networks easily.

Your argument is valid though, that without ssh it somewhat defeats the purpose of allowing to do the initial setup headless too.

vfricke, 2023-09-11:
So, what is the solution to the problem? Will you update the standard image to allow ssh connections per default?

vfricke, 2023-09-13:
Yes. Thank you.

closed 2023-09-13 by vfricke