nymea/berrylan #85: Is Berrylan still active / supported?

ballen4705, 2023-12-31:
I am looking for a framework to configure WiFi for Raspberry Pi IOT devices (based on the Compute Module 4).  Berrylan looks like a good solution.  I have two questions:

- Is it still actively supported?  For example does it work correctly with the latest 64-bit Bookwork release?

- Do the iPhone and Android apps support the latest devices?  I saw some reports that the Android app only works correctly on older Android versions.

Thanks for your work on this.  If it is current and supported, it seems like a great solution for IOT configuration.

Cheers,
   Bruce

mzanetti, 2023-12-31:
nymea-networkmanager, the service on the rpi doing the actual work, should work fine on the latest bookworm images and also is actively maintained as it's also used in the nymea system. The app side of it is also still maintained in nymea-app, however, the berrylan app didn't see any updates in quite a while and its unsure whether it will still be updated due to the lack of contributors.

ballen4705, 2023-12-31:
Thanks for the quick and helpful reply!

When you write, "... the app side of it is also still maintained in nymea-app...", does that mean that in addition to the (uncertain future) berrylan app, there is a different (nymea-app) which can serve the same purpose?  Meaning that there is no need for the berrylan app?

If that is correct, is the nymea-app available both for iOS/iPhone and Android devices?

mzanetti, 2023-12-31:
Sort of... Yes, nymea-app contains the same code as the berrylan app and is maintained in the app stores. However, nymea-app uses this to set up nymea systems.. So after the wifi connection is up, it will try to connect to the nymea service on the device. Using that on a "berrylan" only raspberry pi, will cause to nymea setup to fail. The wifi itself should come up... So if this is just for you, sure, feel free to use it. If you want to set this up for other people, this workflow probably isn't user friendly enough to be what you're looking for.