nymea/berrylan #86: Berrylan quick-start instructions need updating, no longer work

ballen4705, 2024-04-28:
The Berryland quick-start instructions no longer work as they should.  See below for a transcript.  The issue seems to be that the the repository 'http://repository.nymea.io <release> Release' does not have a Release file.

=====================================================

```
ballen@rp:~ $ echo "deb http://repository.nymea.io <release> rpi" | sudo tee /etc/apt/sources.list.d/nymea.list
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-key A1A19ED6
deb http://repository.nymea.io <release> rpi
Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
Executing: /tmp/apt-key-gpghome.FUEQx5ipmm/gpg.1.sh --keyserver keyserver.ubuntu.com --recv-key A1A19ED6
gpg: key 457A6EE4A1A19ED6: public key "nymea GmbH <developer@nymea.io>" imported
gpg: Total number processed: 1
gpg:               imported: 1

ballen@rp:~ $ sudo apt-get update
Hit:1 http://deb.debian.org/debian bullseye InRelease
Hit:2 http://security.debian.org/debian-security bullseye-security InRelease      
Hit:3 http://archive.raspberrypi.org/debian bullseye InRelease                    
Hit:4 http://deb.debian.org/debian bullseye-updates InRelease                     
Ign:5 http://repository.nymea.io <release> InRelease        
Err:6 http://repository.nymea.io <release> Release
  404  Not Found [IP: 92.205.59.180 80]
Reading package lists... Done
E: The repository 'http://repository.nymea.io <release> Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

ballen@rp:~ $ sudo apt-get install nymea-networkmanager dirmngr
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package nymea-networkmanager
ballen@rp:~ $ 

```

t-mon, 2024-04-29:
The `rpi` section has been deprecated, the docs are outdated indeed, need to fix that.

You should be able to add the repository using this:

    echo -e "deb http://repository.nymea.io $(lsb_release -s -c) main" | sudo tee /etc/apt/sources.list.d/nymea.list
    
    sudo wget -O /etc/apt/trusted.gpg.d/nymea.gpg https://repository.nymea.io/nymea.gpg

ballen4705, 2024-04-29:
Thanks, I'll test it later this week.  I see that you've already updated README.md, so from my point of view, the issue can be marked as closed. If I find a problem I'll reopen it in this thread.

closed 2024-04-29 by t-mon