nymea/berrylan #88: mode = always not working

caspersauce, 2024-11-17:
I used the image from this website about 2 weeks ago, and was able to successfully use the iPhone BerryLan app to connect to bluetooth on a Raspberry Pi W 2. I then brought the raspberry pi home a while later and noticed the bluetooth doesn't broadcast. Found a post on here that changing mode to always will make it broadcast again.

This is my conf file:

[General]
Mode=always
Timeout=60
AdvertiseName=nymea-nm
ForceFullName=false
PlatformName=nymea-pi
ButtonGpio=-1
ButtonActiveLow=false
DBusBusType=system

It never broadcasts. It hasn't since the original configuration. 

How do I get it to broadcast again so I can configure the device at a different site?

caspersauce, 2024-11-21:
Is this repository no longer maintained? Maybe that is why it doesn't work properly?

t-mon, 2024-11-21:
This app is actually no longer maintained, the related [nymea-networkmanager](https://github.com/nymea/nymea-networkmanager) doing the work on the pi side is. Either way, the app should still continue to work, since this issue does not seem to be related with the app.

Can you provide the logs of your nymea-networkmanager running on the pi, that should give a hint why it is not working

`journalctl -f -u nymea-networkmanager`

What does `hciconfig -a` give you?

Which image did you use?

caspersauce, 2024-11-22:
that journalctl command just hangs indefinitely... is it correct?

I just did a fresh install and I removed the "-f" argument and took the last lines from the output:

```
Nov 22 23:28:20 nymea systemd[1]: Stopping nymea-networkmanager.service - Daemon for nymea to configure wifi network using a Bluetooth LE connection...
Nov 22 23:28:21 nymea systemd[1]: nymea-networkmanager.service: Deactivated successfully.
Nov 22 23:28:21 nymea systemd[1]: Stopped nymea-networkmanager.service - Daemon for nymea to configure wifi network using a Bluetooth LE connection.
```

This is the last time it ever reports. Even If I restart the pi it never comes back. I assume that is why the the always argument isn't doing anything.

hciconfig -a gives this output:

```
hci0:   Type: Primary  Bus: UART
        BD Address: B8:27:EB:DE:8E:E3  ACL MTU: 1021:8  SCO MTU: 64:1
        UP RUNNING 
        RX bytes:3433 acl:0 sco:0 events:289 errors:0
        TX bytes:50197 acl:0 sco:0 commands:289 errors:0
        Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
        Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
        Link policy: RSWITCH SNIFF 
        Link mode: PERIPHERAL ACCEPT 
        Name: 'nymea'
        Class: 0x000000
        Service Classes: Unspecified
        Device Class: Miscellaneous, 
        HCI Version: 4.2 (0x8)  Revision: 0x412
        LMP Version: 4.2 (0x8)  Subversion: 0x2209
        Manufacturer: Broadcom Corporation (15)
```

I used the October 29th image link from this repository readme

caspersauce, 2024-11-23:
After reading the journalctl I decided to do the obvious and start the service again. This works to enable the bluetooth connection again. 

The commands do not persist through reboot. How do I get the service to always be enabled?

```
sudo systemctl enable nymea-networkmanager
sudo systemctl start nymea-networkmanager
```

t-mon, 2024-11-25:
maybe `systemctl status nymea-networkmanager` gives you a hint why it is not starting automatically.

The last 2 commands you posted should actually do the job.

t-mon, 2024-11-28:
I was able to reproduce the issue, something is not working any more with the service, will check what's wrong with it...

caspersauce, 2024-11-29:
Thank you! I'm glad you were able to reproduce it

caspersauce, 2024-12-04:
Any update on getting the service to run at boot?

t-mon, 2024-12-04:
Looks like the system is not able to finish the boot process because `userconfig.service` is blocking the boot process. 

```
root@nymea:~# systemctl list-jobs
JOB UNIT                                 TYPE  STATE  
92  systemd-update-utmp-runlevel.service start waiting
1   graphical.target                     start waiting
115 userconfig.service                   start running
122 nymea-networkmanager.service         start waiting
2   multi-user.target                    start waiting
```

You actually need to plug in a keyboard and a display in order to finish the process by configuring the user as raspbian seems to want you to do.

You can more information here: https://github.com/RPi-Distro/userconf-pi

If you don't want to do that because you are able to change the default password on your own with `passwd` you just can disable the service followed by a reboot. 

```bash
sudo systemctl disable userconfig.service
sudo reboot
``` 

That should fix the issue the service is not starting, because the boot process was not able to start.

closed 2024-12-04 by t-mon

mzanetti, 2024-12-06:
FWIW, that's what I tried to fix this one: https://github.com/nymea/nymea-networkmanager/pull/60